iRobotEducation/create3_docs #24: Electrical TOC not rendering properly

alsora, 2022-02-17:
[image]

This is the table of content in the Hardware -> Electrical page.
Only one button renders properly

alsora, 2022-02-17:
I usually use Chromium on Ubuntu 20.04.
It does not render in standard Google Chrome either.

I tired with firefox and it renders properly there.

shamlian, 2022-02-17:
Interesting; I am using Chrome on Windows 10 (v98.0.4758.82) and it's rendering. Can you go [here ](https://unicode.org/emoji/charts-12.1/emoji-ordering-rules.txt)and tell me what you see?

shamlian, 2022-02-17:
I'm sorry that this seems like a system configuration problem, not a problem with the docs site.

alsora, 2022-02-17:
For what is worth I see the same thing in Chrome, version 97.0.4692.71

shamlian, 2022-02-17:
I am open to removing emoji from the docs site, or taking them out of the TOC and adding them to the body with a note that they may not render if you're not up to the most recent Unicode.

closed 2022-02-18 by shamlian

shamlian, 2022-02-18:
I think maybe we should decide on a minimum Unicode version to support. Here are the release dates: https://unicode.org/history/publicationdates.html

shamlian, 2022-02-18:
@alsora just for our knowledge, can you see the following:
Added in Unicode 12: 🦩
Added in Unicode 11: 🦡
Added in Unicode 10: 🦔
Added in Unicode 9: 🦐
Added in Unicode 8: 🦄
Added in Unicode 7: 🐿

PR #25 (2022-02-18): Remove Unicode circles and fix yellow indicator text
Removed colored circles since they are from different versions of the Unicode standard and also are not guaranteed to properly render. Also fixed text on yellow indicator. This fixes #24