iRobotEducation/create3_docs #71: Form validation on ROS namespace -> boot loop

carlsondc-ceva, 2022-06-01:
It appears that if you set the ROS namespace to a string containing a "." character, the platform gets stuck in a state where it doesn't fully boot up.  [This is an invalid name](http://design.ros2.org/articles/topic_and_service_names.html) (my bad), and it should probably be rejected by the form.

If such a name is submitted:
1. The light ring spins endlessly
2. The logs, as retrieved with `wget http://192.168.186.2/logs-raw` over the USB interface show the following, repeated roughly every 10 seconds:
```
Oct 26 23:06:55 iRobot-8679B9B5E45344FEB1A5A524C62B5BDB user.notice create-platform: Aborted
Oct 26 23:06:55 iRobot-8679B9B5E45344FEB1A5A524C62B5BDB user.notice create-platform: Server 'create-platform' crashed with exit code 134. Respawning..
Oct 26 23:06:56 iRobot-8679B9B5E45344FEB1A5A524C62B5BDB daemon.notice network[1478]: [INFO] (src/wifi_control.c:1813) wlan monitor event - CTRL-EVENT-SCAN-RESULTS
Oct 26 23:06:56 iRobot-8679B9B5E45344FEB1A5A524C62B5BDB daemon.notice network[1478]: [INFO] (src/wifi_control.c:1813) wlan monitor event - CTRL-EVENT-NETWORK-NOT-FOUND
Oct 26 23:07:00 iRobot-8679B9B5E45344FEB1A5A524C62B5BDB user.notice ntpd: ntpd: bad address '1.irobot.pool.ntp.org'
Oct 26 23:07:01 iRobot-8679B9B5E45344FEB1A5A524C62B5BDB daemon.notice network[1478]: [INFO] (src/wifi_control.c:1813) wlan monitor event - CTRL-EVENT-SCAN-STARTED
Oct 26 23:07:02 iRobot-8679B9B5E45344FEB1A5A524C62B5BDB user.notice create-platform: [WARN] [1635289622.000253089] [rcl]: Namespace not remapped to a fully qualified name (found: /create3.1)
Oct 26 23:07:02 iRobot-8679B9B5E45344FEB1A5A524C62B5BDB user.notice create-platform: [ERROR] [1635289622.001096178] [rcl]: Failed to parse global arguments
Oct 26 23:07:02 iRobot-8679B9B5E45344FEB1A5A524C62B5BDB user.notice create-platform: terminate called after throwing an instance of 'rclcpp::exceptions::RCLInvalidROSArgsError'
Oct 26 23:07:02 iRobot-8679B9B5E45344FEB1A5A524C62B5BDB user.notice create-platform:   what():  failed to initialize rcl: Couldn't parse remap rule: '-r __ns:=/create3.1'. Error: Expected lexeme type 1, got 22 at index 14, at /home/ubuntu/workspace/.conan_homes/mt7688/.conan/data/ros2-core/galactic.irb
```

I was able to get out this state by using curl to change the value, then power-cycle the robot:
```curl -X POST -d "ros_domain_id=${ros_domain_id}&ros_namespace=${ros_namespace}&rmw_implementation=${rmw_implementation}" "http://192.168.186.2/ros-config-save-main" ```
where `ros_domain_id`, `ros_namespace`, and `rmw_implementation` are the values you would have submitted through the web interface.

For reference, the USB setup directions for a Raspberry Pi 4 are [here](https://iroboteducation.github.io/create3_docs/setup/pi4/)

shamlian, 2022-06-01:
Thank you for reporting this bug and a short-term mitigation. We will add it to issue tracker to fix in a subsequent release.

shamlian, 2022-09-14:
Should be solved in G.3.1.