iRobotEducation/create3_docs #96: Create 3 Getting Started Manual bad Python Web Playground link

jomoengineer, 2022-06-30:
In the "Create 3 Getting Started Manual", on page 6 there is a link listed for Python Web Playground titled "Visit the iRobot Education Learning Library". However, when clicking on the link it produces a 404 Error:
```
Oops! We can't seem to find the page you're looking for.
```

Link that fails:
[https://edu.irobot.com/learning-library/create-3-python-web-playground](url)

Perhaps this is the correct link:
[https://edu.irobot.com/learning-library/create-3-web-playground-for-python](url)

shamlian, 2022-06-30:
@jomoengineer Thanks for the report! We'll fix it soon.

Rebecca-iRobot, 2022-06-30:
Redirected! Thank you for your help @jomoengineer!

Rebecca-iRobot, 2022-06-30:
Resolved #96

closed 2022-06-30 by shamlian