iRobotEducation/create3_docs #144: makes available a odom -> base_link publish TF parameter

gbr1, 2022-08-09:
**Is your feature request related to a problem? Please describe.**
If you use SLAM, you have a double odom tf because you have a localization odom (the one wanted)

**Describe the solution you'd like**
a parameter that allows to disable this transformation

**Additional context**
this is useful and other packages use a similar convention

jmm-slamcore, 2022-10-03:
Agree with the above, a parameter to disable `odom->base_footprint` and `odom->base_link` TF publishing would be useful so that other SLAM systems can publish the transforms instead.

shamlian, 2023-01-30:
A parameter has been added in [G.4.1](https://edu.irobot.com/create3/firmware/G.4.1) to address this feature request; please add the following to the parameters yaml on the [application configuration](https://iroboteducation.github.io/create3_docs/webserver/application/#application-ros-2-parameters-file) page on the webserver (the parameter is read only at start and can not be dynamically modified while running):
```
robot_state:
  ros__parameters:
    publish_odom_tfs: false
```

jmm-slamcore, 2023-01-31:
Awesome thanks!

shamlian, 2023-02-27:
Added to H.1

closed 2023-02-27 by shamlian