iRobotEducation/create3_docs #184: Routing ROS2 traffic through an SBC

roni-kreinin, 2022-08-30:
I am looking to set up the TurtleBot 4 to only use WiFi on its RPi4, and for the Create 3 to only use the `usb0` interface. For this to work we need to route Create 3 ROS2 traffic through the RPi4's `wlan0` interface so that an external PC can see the topics/services/etc. I have an open ticket (https://github.com/turtlebot/turtlebot4/issues/26) where we have been trying to get this working but so far we haven't found solution.

I mentioned [here](https://github.com/turtlebot/turtlebot4/issues/26#issuecomment-1231915979) that I was able to get discovery working using the Fast-DDS discovery server in a setup where I used a second RPi4 to emulate the Create 3. Each PC in this setup is required to export the `ROS_DISCOVERY_SERVER` envar to set the IP/PORT of the server. For this to work with the Create 3, we would need to add a way to set this envar in the webserver.

There is still the issue of routing the unicast packets that I am still looking in to. Any help is appreciated.

alexswerner, 2022-09-13:
Hi, I want to second this request. In our situation using the Creates Wi-Fi has proven problematic. We bought 15 Create3 (Turtlebot 4) for use in classes. However, using 2.4Ghz Wi-Fi on a university campus does not seem to give us the reliablity we need. So the best option seems to be to access the base though the `usb0` interface of the raspberry pi. The raspberry pis Wi-Fi is very stable in comparison.

As @roni-kreinin described, we want to disable the Creates Wi-Fi entirely. However the static network configuration of the Create does not allow its ROS domain participants to reach any other participants beyond it's local network (192.168.186.0/24) because not default route is set. One option to address this would be to enable the dhcp client on the usb network interface. Then the raspberry pi could provide the base with the necessary network information.

I would be happy to test a firmware.

alexswerner, 2022-10-17:
@shamlian is there any way we can contribute to the development here? Like this, some indended uses of the robot are not possible due to the lack of reliability.

iRobotEdu, 2022-10-18:
Hi @alexswerner, would you mind reaching out to education@irobot.com with this inquiry? Many issues/comments in this forum are answered by volunteers, and we'd like to get an official request in place.

shamlian, 2023-01-30:
This should be fixed in [G.4.1](https://edu.irobot.com/create3/firmware/G.4.1); please download it and let us know.

shamlian, 2023-02-27:
Fixed in H.1

closed 2023-02-27 by shamlian