iRobotEducation/create3_docs #221: root access?

MAVProxyUser, 2022-10-11:
The drop bear service is disabled unless development mode is set. These units ship as production mode. 

How can we gain ssh access?

alsora, 2022-10-11:
Hi @MAVProxyUser, you can't ssh into the robot.

The robot communicates with the external world using ROS 2 or BLE so you can use one of them to command the robot or listen to its sensor data.
Could you give us more details on your specific use-case and why you would need ssh access? I'm sure we can find a solution that does not involve that.

alsora, 2022-10-11:
I'm sorry, our product security team doesn't allow us to enable SSH on the Create 3 robot.

MAVProxyUser, 2022-10-11:
They don't have a choice but to comply with GPL. So please open up your linux sub system. You can get sued if you chose not to. Please don't break the law. You can choose to not share your ssh access, you however can NOT choose to not share your linux subsystem. 
"Do everyone a favor and don't break the law -- if you use busybox, comply with the busybox license by releasing the source code with your product." - https://web.archive.org/web/20070126122736/https://busybox.net/shame.html

Please advise. 
https://www.gnu.org/licenses/gpl-faq.en.html#InstInfo
"it does require redistribution to include scripts used to control compilation and installation of the executable with the complete and corresponding source code"

You distribute GPL binaries, you must share the source. Attempting to keep people out of your Linux subsystem has some ramifications. 
https://gpl-violations.org/faq/sourcecode-faq/
"For each and every version of the executable program, you have to release the precisely corresponding version of the complete corresponding source code.

So if you have distributed ten different versions of firmware for an embedded product, and this firmware contains GPL licensed software, then you need to release ten different source code packages, each one corresponding for each executable version."

MAVProxyUser, 2022-10-11:
@alsora please check in with the legal team re: GPL obligations. They are non-optional.

hotelzululima, 2022-10-11:
Hi @Alsora,
    by your using GPL code to build your product you dont get the choice to keep the source code private, you are using other folks copylefted code that you did specifically NOT develop  and simply used instead. Using GPL licensed code comes  with a set of legal obligations.  Either live up to your legal responsibilities, OR pull your product off the market till you  eliminate every last bit of GPL licensed code linked into your product OR get sued & have this issue blasted wide across every social media that you have stolen GPL code to use in your product.  

       your choice
       hzl

MAVProxyUser, 2022-10-11:
Subsequently referenced here for security and legal to understand they are related, but intertwined problems. https://github.com/iRobotEducation/create3_docs/issues/222

iRobotEdu, 2022-10-13:
Upon further review, we do have a process where you can request source code licensed under GPL-style licenses. You would need to: (1) send us the name of the specific Create 3 open source components that you would like the source code for, (2) include $5.00 for each component, and (3) provide your return mailing address to:

iRobot Corporation 
Attn: Open Source – Legal Department
8 Crosby Drive
Bedford, MA 01730

We will then provide the requested source code on a durable medium for you.

MAVProxyUser, 2022-10-13:
Wow @iRobotEdu look at you guys... <insert *trash* .gif here>
https://media.giphy.com/media/xT39D7O9Xj1JqKq5i0/giphy.gif

I want you to give me a list of very component one the system FIRST, by running the 'legal-info' script. You've created a chicken / egg scenario out of bad faith. As if you can't just put the components on GitHub... charging me $5 per component for an unnamed amount of components is a dirty way to treat the open source community that supports you. 

https://github.com/iRobotEducation/create3_docs/issues/222

hotelzululima, 2022-10-13:
ok its time to set a twitter bot specifying that this robot vendor is selling stolen GPL code...  

I am tired of bullshit like this from yet another set of  GPL thieves..
see how some very negative publicity affects this thieving companies' bottom line.
      

    bye 'felecia

MAVProxyUser, 2022-10-15:
Oh man... the crows sure came home to roost here eh!? Have the security and legal team re-watch this talk from Patrick Doyle in full, from this point on: https://youtu.be/t54T9uemiRU?t=1662

iRobotEdu, 2022-10-21:
@MAVProxyUser, we appreciate your fervor and passion for coding with the Create® 3 educational robot. The open source components of the open-embedded build recipe for our Linux distribution are available here:
1. https://github.com/iRobotEducation/meta-mediatek-mt7688
2. https://github.com/iRobotEducation/meta-irobot-mt
3. https://github.com/iRobotEducation/meta-irobot-common

MAVProxyUser, 2022-10-21:
I appreciate you doing the right thing. Thank you kindly for your time.