iRobotEducation/create3_docs #226: root access + 3rd usb ethernet adapter support on USBC

MAVProxyUser, 2022-10-14:
Sadly you prematurely closed my request for root access. https://github.com/iRobotEducation/create3_docs/issues/221

@alsora suggested "Could you give us more details on your specific use-case and why you would need ssh access? I'm sure we can find a solution that does not involve that."

While I wait for the legal department to learn how GPL works (https://github.com/iRobotEducation/create3_docs/issues/225), lets walk through your scenario above:
I would like to plug a USB ethernet device into the USBC port, and use it to connect the bot to a local ethernet network. 
CONFIG_USB_USBNET should handle it... sadly because you don't share the kernel a) I can't see the kernel config. b) even if you do have it enabled you've given me no way to act as the network administrator of my own linux box inside the Create 3 platform. As such I can't manually use ifconfig to assign an IP address range. 

So... short of telling me "no, you can't use USB ethernet cuz our linux is a black box", your option is to ensure that CONFIG_USB_USBNET is compiled in the next kernel update? and to expose something in the web interface for me to edit the options?

MAVProxyUser, 2022-10-15:
Oh man... the crows sure came home to roost here eh!? Have the security and legal team re-watch this talk from Patrick Doyle in full, from this point on: https://youtu.be/t54T9uemiRU?t=1662

Jayden-F, 2023-02-06:
I would greatly appreciate root access to the create. I would like configure the chrony service on the create3. So I do not need to restart the robot numerous in the hope it might time sync to the connected raspberry pi.

Sadly the solution in the documentation is not effective. 

>8. Note that if there is a large jump in the time, the Create® 3 may not accept it until its next reboot. This can be verified by checking the Create® 3 robot's log for a line like
user.notice ntpd: ntpd: reply from 192.168.186.3: delay ### is too high, ignoring
If this happens, simply restart the robot (not just the application) via the webserver over the USB network connection.

MAVProxyUser, 2023-02-06:
They won’t give it out… they are afraid. I forced them to share the GPL buildroot, but they won’t even give instructions to compile it and flash your own firmware. Soooo open!!!

Jayden-F, 2023-02-06:
So I am forced to play reboot roulette to time sync... damn.

MAVProxyUser, 2023-02-06:
Here are the open source repos that they were reluctant to give out. 
https://github.com/iRobotEducation/meta-irobot-common
https://github.com/iRobotEducation/meta-mediatek-mt7688
https://github.com/iRobotEducation/meta-irobot-mt/

Until @irobot-ros @iRobotEdu stops ignoring the request for more documentation, yes. 
https://github.com/iRobotEducation/meta-irobot-mt/issues/1