iRobotEducation/create3_docs #309: Create 3 returns invalid accelerometer values in BLE mode

shamlian, 2023-01-13:
[root-robot-ble-protocol](https://github.com/iRobotEducation/root-robot-ble-protocol#device-16---accelerometer) requires accelerometer readings to be inclusive of gravity, but current versions of Create 3 firmware return values from /imu which are in the robot's coordinate frame and remove gravity. Please fix it, me!

shamlian, 2023-01-30:
This should be fixed in [G.4.1](https://edu.irobot.com/create3/firmware/G.4.1); please download it and let us know!

closed 2023-01-31 by shamlian

LautaroRinaldi, 2024-05-22:
Is there a fix for Humble?

shamlian, 2024-05-22:
The only version of Humble firmware that should have been affected by this is H.0.0, since H.1.0 has feature parity with G.4.3. Are you experiencing a problem, and if so, what version of firmware are you running? We recommend running the latest.