iRobotEducation/create3_docs #312: Cliff sensor and touch sensor problems

scottcandy34, 2023-01-17:
I have create3 robot Firmware version (G.3.1) the current cliff sensor does not trigger an event based on the Bluetooth docs here https://github.com/iRobotEducation/root-robot-ble-protocol#command-0---cliff-event although it does trigger on the touch sensor event. Very strange since the documentation does not describe this possibility here https://github.com/iRobotEducation/root-robot-ble-protocol#command-0---touch-sensor-event

I have found this issue while programing in https://python.irobot.com your current documentation does not cover this I don't know if it was an oversight or what, but can this be fixed so its on its own event posting instead of the touch sensor event or just update the documentation at least. 

The sensors values are on byte 7 with hex values of 1, 2, 4, 8 from the left to the right and of course a combination of them when multiple are triggered.

Here is a link to the issue also posted in the irobot-edu-python-sdk
https://github.com/iRobotEducation/irobot-edu-python-sdk/issues/12

scottcandy34, 2023-01-17:
Ok thx

shamlian, 2023-01-30:
This should be fixed in [G.4.1](https://edu.irobot.com/create3/firmware/G.4.1); please download it and let us know!

shamlian, 2023-02-08:
Confirmed this is fixed

closed 2023-02-08 by shamlian