iRobotEducation/create3_docs #316: Solid red light and navigation problem after update create3 to G.3.1

JasmineRosselliSUPSI, 2023-01-20:
!**This problem has also been reported in turtlebot/turtlebot4 community**! (https://github.com/turtlebot/turtlebot4/issues/90)

I report here the message I post yesterday on the turtlebot/turtlebot4 community, since I saw that many others have reported the same problem in this community too. This post was originally related to another user's issue.
___
" 
I recently had a similar problem after updating the create3 (but only a few weeks leater, so it my not be the cause) to the newer version (G.3.1). I have the versions ubuntu 20.04 Focal and ROS2 Galactic installed on the Rpi of my turtlebot4.

The similarities are:
- solid red light and turtlebot4 unable to react to any stimulus
- navigation issues
- the turtlebot4 CAN be driven using `ros2 run teleop_twist_keyboard teleop_twist_keyboard`
- the turtlebot4 CAN create maps (`ros2 launch turtlebot4_navigation slam_sync.launch.py`)
- the turtlebot4 CANNOT be driven using `ros2 launch turtlebot4_navigation nav_bringup.launch.py slam:=off localization:=true map:=<map_name>.yaml`

One more problem I have:
- /docking and /undocking don't work from script or through command line (action send_goals) -> BUT the robot is actually able to perform /dock and /undock using create3 physical buttons.

If run from script or terminal, it returns message:
'_Goal_rejected_' 

If turtlebot4 is not on docker station, always respond with `Goal_rejected` but sometimes perform dock anyway.

(Edited):
Here I leave the terminal response when I try to perform /dock or /undock from script. Despite error message, /dock was performed.

`$ ros2 run navigation_test test_crash`
_[INFO] [1674208103.292496984] [basic_navigator]: Docking before initialising pose
[INFO] [1674208103.294937391] [basic_navigator]: Docking...
[ERROR] [1674208103.304079597] [basic_navigator]: Dock goal rejected
Traceback (most recent call last):
  File "/home/<my_raspberry_name>/turtlebot4_ws/install/navigation_test/lib/navigation_test/test_crash", line 11, in <module>
    load_entry_point('navigation-test==0.0.0', 'console_scripts', 'test_crash')()
  File "/home/<my_raspberry_name>/turtlebot4_ws/install/navigation_test/lib/python3.8/site-packages/navigation_test/test_dock_undock_crash.py", line 21, in main
    navigator.dock()
  File "/home/<my_raspberry_name>/turtlebot4_ws/install/turtlebot4_navigation/lib/python3.8/site-packages/turtlebot4_navigation/turtlebot4_navigator.py", line 214, in dock
    while not self.isDockComplete():
  File "/home/<my_raspberry_name>/turtlebot4_ws/install/turtlebot4_navigation/lib/python3.8/site-packages/turtlebot4_navigation/turtlebot4_navigator.py", line 238, in isDockComplete
    if self.dock_result_future is None or not self.dock_result_future:
AttributeError: 'TurtleBot4Navigator' object has no attribute 'dock_result_future'_

---
Here some more details
--- 

At first I did the update of create3 to solve nav2 a problem (often amcl_pose parameter wasn't publishing and the only way to fix that was to reboot the robot multiple times).
For some weeks this worked well.

BUT! 

I now face too with navigation issues, with solid red light and turtlebot4 unable to receive or send commands, even from physical buttons. To momentarily fix the issue, the only thing that worked was to take off the battery and reboot the system.

I also started to notice problems at action and service level, which seem to be cumulative. The more action and service servers are compromised, the more others follow.

From create3 log i receive both **retcode -3** and **retcode -1** errors. I already tried to install older version of create3 and back to new, but in neither case worked.

If I test turtlebot4 behavior running `ros2 run turtlebot4_tests ros_tests`,
I get stuck at tests 3 and 4.
The messages I get are basically the same (here some examples):

_- [WARN] [1674139895.970490189] [turtlebot4_ros_tests]: Ignoring unexpected goal response. There may be more than one action server for the action '/drive_distance'_ 

_- [WARN] [1674139974.115775685] [turtlebot4_ros_tests]: Ignoring unexpected result response. There may be more than one action server for the action '/dock'_ 

Test 3 actually works, beside the WARN messages, but on test 4 the turtlebot4 performs undock and then crash.

When I exit the test terminal, the following message error occurs:

_Exception ignored in: <function ActionClient.__del__ at 0xffff8a07c940>
Traceback (most recent call last):
  File "/opt/ros/galactic/lib/python3.8/site-packages/rclpy/action/client.py", line 588, in __del__
  File "/opt/ros/galactic/lib/python3.8/site-packages/rclpy/action/client.py", line 581, in destroy
rclpy._rclpy_pybind11.InvalidHandle: cannot use Destroyable because destruction was requested_

Nodes and packages not related to navigation work fine.

Could this be a problem related to the create3 base or to the turtlebot4?
Has anyone been able to solve this problem?
"
___
I tried to fix the 'ddsi_udp_conn_write retcode -3' issue following the advice at https://answers.ros.org/question/390859/ddsi_udp_conn_write-failed/, but it didn't fix my issue.

Any kind of help will be greatly appreciated!

shamlian, 2023-01-23:
The red light is interesting -- any chance you have collected logs from the robot in this state (access /logs-raw on the robot's webserver)? Without a payload, this is usually an indication that the Create 3 robot's application is using a lot of resources (though I am not sure if TB4 overloads this light).

JasmineRosselliSUPSI, 2023-01-24:
Thank you for the quick response!

I have collected the current log, both for turtlebot4 connected and non-connected. Unfortunately I don't have any recorded at the time of the crash.
___
I am sorry for the extremely long message, I hope this could be helpful.
___
Here a part of the log collected just after the boot of the robot (Raspberry of the turtlebot4 CONNECTED).

Jan 24 07:54:36 iRobot-CE577DC244DA42909FA6E7428AB4BF09 user.notice create-platform: 1674546876.462469 [0]    nodes_0: ddsi_udp_conn_write to udp/10.17.11.53:59331 failed with retcode -1
Jan 24 07:54:36 iRobot-CE577DC244DA42909FA6E7428AB4BF09 user.notice create-platform: 1674546876.463238 [0]    nodes_0: ddsi_udp_conn_write to udp/10.17.11.53:59331 failed with retcode -1
Jan 24 07:54:36 iRobot-CE577DC244DA42909FA6E7428AB4BF09 user.notice create-platform: 1674546876.476428 [0]    nodes_0: ddsi_udp_conn_write to udp/10.17.11.53:59331 failed with retcode -1
Jan 24 07:54:36 iRobot-CE577DC244DA42909FA6E7428AB4BF09 user.notice create-platform: 1674546876.476814 [0]    nodes_0: ddsi_udp_conn_write to udp/10.17.11.53:59331 failed with retcode -1

[...]

Jan 24 07:54:37 iRobot-CE577DC244DA42909FA6E7428AB4BF09 user.notice create-platform: 1674546877.103739 [0]        tev: ddsi_udp_conn_write to udp/10.17.11.53:59331 failed with retcode -1
Jan 24 07:54:37 iRobot-CE577DC244DA42909FA6E7428AB4BF09 user.notice create-platform: 1674546877.103859 [0]    nodes_0: ddsi_udp_conn_write to udp/10.17.11.53:59331 failed with retcode -1
Jan 24 07:54:37 iRobot-CE577DC244DA42909FA6E7428AB4BF09 user.notice create-platform: 1674546877.117135 [0]    nodes_0: ddsi_udp_conn_write to udp/10.17.11.53:59331 failed with retcode -1

[...]

Jan 24 07:54:37 iRobot-CE577DC244DA42909FA6E7428AB4BF09 user.notice create-platform: 1674546877.645996 [0]    nodes_0: ddsi_udp_conn_write to udp/10.17.11.53:59331 failed with retcode -1
Jan 24 07:54:37 iRobot-CE577DC244DA42909FA6E7428AB4BF09 user.notice create-platform: 1674546877.646713 [0]    nodes_0: ddsi_udp_conn_write to udp/10.17.11.53:59331 failed with retcode -1
Jan 24 07:54:37 iRobot-CE577DC244DA42909FA6E7428AB4BF09 user.notice create-platform: 1674546877.649979 [0]        tev: ddsi_udp_conn_write to udp/10.17.11.53:42047 failed with retcode -1
Jan 24 07:54:37 iRobot-CE577DC244DA42909FA6E7428AB4BF09 user.notice create-platform: 1674546877.650105 [0]        tev: ddsi_udp_conn_write to udp/10.17.11.53:44418 failed with retcode -1
Jan 24 07:54:37 iRobot-CE577DC244DA42909FA6E7428AB4BF09 user.notice create-platform: 1674546877.650162 [0]        tev: ddsi_udp_conn_write to udp/10.17.11.53:57183 failed with retcode -1
Jan 24 07:54:37 iRobot-CE577DC244DA42909FA6E7428AB4BF09 user.notice create-platform: 1674546877.650216 [0]        tev: ddsi_udp_conn_write to udp/10.17.11.53:57808 failed with retcode -1
Jan 24 07:54:37 iRobot-CE577DC244DA42909FA6E7428AB4BF09 user.notice create-platform: 1674546877.650270 [0]        tev: ddsi_udp_conn_write to udp/10.17.11.53:59331 failed with retcode -1
Jan 24 07:54:37 iRobot-CE577DC244DA42909FA6E7428AB4BF09 user.notice create-platform: 1674546877.660434 [0]    nodes_0: ddsi_udp_conn_write to udp/10.17.11.53:59331 failed with retcode -1

[...]

Jan 24 07:54:44 iRobot-CE577DC244DA42909FA6E7428AB4BF09 user.notice create-platform: 1674546884.093989 [0]    nodes_0: ddsi_udp_conn_write to udp/10.17.11.53:59331 failed with retcode -1
Jan 24 07:54:44 iRobot-CE577DC244DA42909FA6E7428AB4BF09 user.notice create-platform: 1674546884.097937 [0]    nodes_0: ddsi_udp_conn_write to udp/10.17.11.53:59331 failed with retcode -1

JasmineRosselliSUPSI, 2023-01-24:
Here a part of the log collected (Raspberry of the turtlebot4 NOT CONNECTED).

Jan 24 08:01:50 iRobot-CE577DC244DA42909FA6E7428AB4BF09 user.notice create-platform: 1674547310.082724 [0]    nodes_0: ddsi_udp_conn_write to udp/10.17.11.53:59331 failed with retcode -1
Jan 24 08:01:50 iRobot-CE577DC244DA42909FA6E7428AB4BF09 user.notice create-platform: 1674547310.085968 [0]    nodes_0: ddsi_udp_conn_write to udp/10.17.11.53:59331 failed with retcode -1
Jan 24 08:01:50 iRobot-CE577DC244DA42909FA6E7428AB4BF09 user.notice create-platform: 1674547310.086608 [0]    nodes_0: ddsi_udp_conn_write to udp/10.17.11.53:57183 failed with retcode -1

[...]

Jan 24 08:01:50 iRobot-CE577DC244DA42909FA6E7428AB4BF09 user.notice create-platform: 1674547310.101435 [0]    nodes_0: ddsi_udp_conn_write to udp/10.17.11.53:59331 failed with retcode -1
Jan 24 08:01:50 iRobot-CE577DC244DA42909FA6E7428AB4BF09 user.notice create-platform: 1674547310.102392 [0]        tev: ddsi_udp_conn_write to udp/10.17.11.53:57808 failed with retcode -1
Jan 24 08:01:50 iRobot-CE577DC244DA42909FA6E7428AB4BF09 user.notice create-platform: 1674547310.102510 [0]        tev: ddsi_udp_conn_write to udp/10.17.11.53:57183 failed with retcode -1
Jan 24 08:01:50 iRobot-CE577DC244DA42909FA6E7428AB4BF09 user.notice create-platform: 1674547310.103954 [0]        tev: ddsi_udp_conn_write to udp/10.17.11.53:57183 failed with retcode -1
Jan 24 08:01:50 iRobot-CE577DC244DA42909FA6E7428AB4BF09 user.notice create-platform: 1674547310.104744 [0]        tev: ddsi_udp_conn_write to udp/10.17.11.53:57183 failed with retcode -1
Jan 24 08:01:50 iRobot-CE577DC244DA42909FA6E7428AB4BF09 user.notice create-platform: 1674547310.116137 [0]    nodes_0: ddsi_udp_conn_write to udp/10.17.11.53:59331 failed with retcode -1
Jan 24 08:01:50 iRobot-CE577DC244DA42909FA6E7428AB4BF09 user.notice create-platform: 1674547310.116527 [0]    nodes_0: ddsi_udp_conn_write to udp/10.17.11.53:59331 failed with retcode -1
Jan 24 08:01:50 iRobot-CE577DC244DA42909FA6E7428AB4BF09 user.notice create-platform: 1674547310.117121 [0]    nodes_0: ddsi_udp_conn_write to udp/10.17.11.53:57183 failed with retcode -1
Jan 24 08:01:50 iRobot-CE577DC244DA42909FA6E7428AB4BF09 user.notice create-platform: 1674547310.117203 [0]    nodes_0: ddsi_udp_conn_write to udp/10.17.11.53:59331 failed with retcode -1
Jan 24 08:01:50 iRobot-CE577DC244DA42909FA6E7428AB4BF09 user.notice create-platform: 1674547310.118614 [0]    nodes_0: ddsi_udp_conn_write to udp/10.17.11.53:59331 failed with retcode -1
Jan 24 08:01:50 iRobot-CE577DC244DA42909FA6E7428AB4BF09 user.notice create-platform: 1674547310.119786 [0]        tev: ddsi_udp_conn_write to udp/10.17.11.53:42047 failed with retcode -1
Jan 24 08:01:50 iRobot-CE577DC244DA42909FA6E7428AB4BF09 user.notice create-platform: 1674547310.131441 [0]    nodes_0: ddsi_udp_conn_write to udp/10.17.11.53:59331 failed with retcode -1

[...]

Jan 24 08:01:50 iRobot-CE577DC244DA42909FA6E7428AB4BF09 user.notice create-platform: 1674547310.181296 [0]    nodes_0: ddsi_udp_conn_write to udp/10.17.11.53:59331 failed with retcode -1
Jan 24 08:01:50 iRobot-CE577DC244DA42909FA6E7428AB4BF09 daemon.notice network[1467]: [INFO] (src/gateway_pinger.c:113) wifi is disconnected
Jan 24 08:01:50 iRobot-CE577DC244DA42909FA6E7428AB4BF09 daemon.notice network[1467]: [INFO] (src/gateway_pinger.c:219) no gateway to ping
Jan 24 08:01:50 iRobot-CE577DC244DA42909FA6E7428AB4BF09 user.notice create-platform: 1674547310.195959 [0]    nodes_0: ddsi_udp_conn_write to udp/10.17.11.53:59331 failed with retcode -1
Jan 24 08:01:50 iRobot-CE577DC244DA42909FA6E7428AB4BF09 user.notice create-platform: 1674547310.196354 [0]    nodes_0: ddsi_udp_conn_write to udp/10.17.11.53:59331 failed with retcode -1
Jan 24 08:01:50 iRobot-CE577DC244DA42909FA6E7428AB4BF09 user.notice create-platform: 1674547310.197720 [0]    nodes_0: ddsi_udp_conn_write to udp/10.17.11.53:57183 failed with retcode -1
Jan 24 08:01:50 iRobot-CE577DC244DA42909FA6E7428AB4BF09 user.notice create-platform: 1674547310.197806 [0]    nodes_0: ddsi_udp_conn_write to udp/10.17.11.53:59331 failed with retcode -1
Jan 24 08:01:50 iRobot-CE577DC244DA42909FA6E7428AB4BF09 user.notice create-platform: 1674547310.198424 [0]    nodes_0: ddsi_udp_conn_write to udp/10.17.11.53:59331 failed with retcode -1
Jan 24 08:01:50 iRobot-CE577DC244DA42909FA6E7428AB4BF09 user.notice create-platform: 1674547310.199644 [0]        tev: ddsi_udp_conn_write to udp/10.17.11.53:42047 failed with retcode -1
Jan 24 08:01:50 iRobot-CE577DC244DA42909FA6E7428AB4BF09 user.notice create-platform: 1674547310.211467 [0]    nodes_0: ddsi_udp_conn_write to udp/10.17.11.53:59331 failed with retcode -1

[...]

Jan 24 08:01:50 iRobot-CE577DC244DA42909FA6E7428AB4BF09 user.notice create-platform: 1674547310.515273 [0]    nodes_0: ddsi_udp_conn_write to udp/10.17.11.53:59331 failed with retcode -1
Jan 24 08:01:50 iRobot-CE577DC244DA42909FA6E7428AB4BF09 user.notice create-platform: 1674547310.515666 [0]    nodes_0: ddsi_udp_conn_write to udp/10.17.11.53:59331 failed with retcode -1
Jan 24 08:01:50 iRobot-CE577DC244DA42909FA6E7428AB4BF09 user.notice create-platform: 1674547310.516261 [0]    nodes_0: ddsi_udp_conn_write to udp/10.17.11.53:57183 failed with retcode -1
Jan 24 08:01:50 iRobot-CE577DC244DA42909FA6E7428AB4BF09 user.notice create-platform: 1674547310.516343 [0]    nodes_0: ddsi_udp_conn_write to udp/10.17.11.53:59331 failed with retcode -1
[... cut ...]
Jan 24 08:14:03 iRobot-CE577DC244DA42909FA6E7428AB4BF09 daemon.notice network[1467]: [INFO] (src/gateway_pinger.c:113) wifi is disconnected
Jan 24 08:14:03 iRobot-CE577DC244DA42909FA6E7428AB4BF09 daemon.warn network[1467]: [WARN] (src/gateway_pinger.c:132) no address count, 4
Jan 24 08:14:03 iRobot-CE577DC244DA42909FA6E7428AB4BF09 user.notice ntpd: ntpd: send failed: Network unreachable
Jan 24 08:14:05 iRobot-CE577DC244DA42909FA6E7428AB4BF09 user.notice create-platform: [INFO] [1674548045.317478534] [system_health]: CPU usage: max 67 [%] mean 41 [%] RAM usage: 26/59 [MB]

JasmineRosselliSUPSI, 2023-01-24:
Here the whole logs:

[messages_turtlebot4_Rpi_CONNECTED.txt](https://github.com/iRobotEducation/create3_docs/files/10487669/messages_turtlebot4_Rpi_CONNECTED.txt)

[messages_turtlebot4_Rpi_NOT_CONNECTED.txt](https://github.com/iRobotEducation/create3_docs/files/10487667/messages_turtlebot4_Rpi_NOT_CONNECTED.txt)

JasmineRosselliSUPSI, 2023-02-01:
I have an update! The turtlebot4 navigation problems seem to be fixed.

---
The SD card of the turtlebot4 unexpectedly broke down and had to be replaced. It has been replaced with a SD with an older turtlebot4 backup. The .git of folders turtlebot4, turtlebot4_robot and irobot_create_msgs have been updated to reach the version just before Humble.

After this procedure it seems that the turtlebot4 navigation works again.
Although an old SD card had already been tested with a previous backup, with inconclusive results (navigation still did not work), it is possible that the .git update action was helpful. It is also possible that a problem with the used ubuntu account occurred with this test, because it turned out that this account have some rviz display issues.

---
Me and my team have some suppositions of the real nature of the navigation error.

- It is possible that the SD card was defective and/or got damaged during one of the multiple forced shutdowns of the device (used to solve the solid red light problems).
- This could have caused knock-on problems, such as memory corruption related to the handling of topics, services, and actions (for example /dock and /undock)
-  **It is possible that navigation problems are then NOT connected directly to the Create3 base!**
   - Warnings and errors of navigation (for example /dock) may have been caused by the inability to receive feedback or response messages from the create3

---
I think that the navigation issue can be considered as fixed.

shamlian, 2023-02-01:
Glad to hear your issue is solved; can I close this ticket, or is there another issue that might need solving on the Create 3 robot?

shamlian, 2023-02-08:
You're quite welcome!

closed 2023-02-08 by shamlian