iRobotEducation/create3_docs #320: Feature request: Cliff Sensor Intensity Value

shamlian, 2023-01-24:
### Discussed in https://github.com/iRobotEducation/create3_docs/discussions/319

<div type='discussions-op-text'>

<sup>Originally posted by **Kola50011** January 24, 2023</sup>
Hey! 
I am working on a project, where we would like to detect a black line on the floor using the 4 cliff sensors. Sadly, this information not published by the ROS API. Is there any way to access such information, other than the _/hazard_detection_ topic? We  would greatly appreciate something similar to the _/ir_intensity_ topic.</div>

Kola50011, 2023-03-21:
Hello, I saw that this feature was removed from H.1.0 - any plans when this might be added? I would really appreciate native support instead of having a workaround and mounting my own, additional sensors to the robot.

shamlian, 2023-03-22:
Hi @Kola50011! Not to worry, this feature is still in development. As soon as there is a planned release, I will let you know via this issue.

fmorshed52, 2023-06-05:
What is the ETA on this feature?

Thank you,

shamlian, 2023-06-07:
It is the team's intention to release this very soon.

PR #420 (2023-07-06): Release G.5.1 / H.2.1
## Changelog
### ROS 2
* Publications
    * Add `/cliff_intensity` topic (fixes #320)
    * Add tf frame for the IMU (called `imu`), to which the `/imu` topic is now connected
* Middleware
    * Allow the use of `;` characters in Fast-DDS Discovery Server configuration (fixes #400)
    * Beta: Allow user to provide XML configuration file for their preferred middleware (fixes #407)
### Core Robot
* Webserver / OS
    * Beta: Allow user full control to edit ntp.conf (fixes #333)
    * Remove erroneous log messages about the robot's lost Wi-Fi connection when it had never be [... cut]
closed 2023-07-06 by shamlian