iRobotEducation/create3_docs #327: Create3 and Raspberry Pi

Jayden-F, 2023-02-06:
Hello, I am encountering difficulties in setting up my TurtleBot4 robot. The main issue is regarding the time sync of the raspberry pi and create. 

SETUP:
I am currently running the robot in an isolated network. The raspberry Pi is set up to source its time from my remote desktop. The create3 requires a ridiculous number of reboots to successfully sync its time with the Raspberry Pi. 

Is there a better solution?

Can we have ssh access to manually configure the chrony service on the create3?

RELATED ISSUE:
https://github.com/turtlebot/turtlebot4_robot/issues/8

Jayden-F, 2023-02-08:
There is obviously this: https://github.com/iRobotEducation/create3_docs/discussions/324, although it would nice to solve the delay issue directly.

shamlian, 2023-02-08:
There are a few moving parts to this request, but I think I can help make the clock sync a little less painful quickly in a beta feature; this should help you and [carlsondc-ceva](https://github.com/carlsondc-ceva) while we work on a better solution. Stand by.

shamlian, 2023-02-13:
Hi there,
In [G.4.3](https://iroboteducation.github.io/create3_docs/releases/g_4_3/), there are now two beta features which may help: first, an option to [restart ntpd](https://iroboteducation.github.io/create3_docs/webserver/restart-ntpd/), which will cause the robot to attempt a time resync immediately, and another page which allows you to [set the time](https://iroboteducation.github.io/create3_docs/webserver/set-datetime/), which may be less useful, except that it shows the current system time on page load, and may make it easier to verify sync.
Hope these help.

Jayden-F, 2023-02-13:
@shamlian Thank you, additionally, as an end users it would be helpful to edit the chrony.conf file on the create3, using a similar process as the "Application ROS2 Parameter File". So that we can configure the create3 to directly source from a local ntp time server.

shamlian, 2023-02-14:
Sure thing. I created a new ticket (#333) for this feature request. I will keep this ticket open until we release these beta features on a Humble build. FWIW I believe the robot should be able to get NTP sources over DHCP as it is now, but I need to double check that is true.

shamlian, 2023-02-27:
Added to H.1

closed 2023-02-27 by shamlian

royito55, 2023-06-28:
As a followup, the Restart ntpd Beta Feature seems to not do anything, or maybe I'm doing something wrong?

Described here, happens in H.1.2

https://github.com/iRobotEducation/create3_docs/issues/414