iRobotEducation/create3_docs #335: Add the option to change the static usb0 subnet

roni-kreinin, 2023-02-21:
Currently it is not possible to run multiple TurtleBot 4's using discovery server with one remote PC because all Create 3's have the same `usb0` IP. The PC cannot create multiple ip routes to `192.168.186.0/24`. Adding the option to change the subnet to a unique value (i.e. `192.168.187.0`) using the webserver should solve this issue. The IP address should probably still be forced to xxx.xxx.xxx.2.

Jayden-F, 2023-03-08:
@shamlian there seems to so many roadblocks for getting multiple turtlebot4 working on a single network. Could you please look into this and let us know if it can be resolved?

shamlian, 2023-03-08:
The team is looking into this (and other network requests) now.

shamlian, 2023-04-17:
To avoid any further delay, I've added this as a beta feature as part of a patch release (G.4.5 / H.1.2).

closed 2023-06-23 by shamlian