iRobotEducation/create3_docs #351: Create3 Fails to run consecutive rotate goals with RCLPY ROS2

Jacob-Friedberg, 2023-03-10:
### Discussed in https://github.com/iRobotEducation/create3_docs/discussions/350

<div type='discussions-op-text'>

<sup>Originally posted by **Jacob-Friedberg** March  9, 2023</sup>
**ATTENTION: To run this code you MUST be on XORG/X11 the pynput library DOES NOT WORK on Wayland**

**System Info**
Operating System: Kubuntu 22.04
Kernel Version: 5.15.0-67-generic (64-bit)
Graphics Platform: X11
Python Version: 3.10.6 (64-bit)

**ROS Info**
ROS_DISTRO=Humble
RMW_IMPLEMENTATION=rmw_cyclonedds_cpp

**Create3 Info**
Firmware Version: H.0.0
ROS 2 Domain ID: 0
RMW_IMPLEMENTATION:rmw_cyclonedds_cpp
Application Ros2 Parameters file:
```
create3_05F8/motion_control:
  ros__parameters:
    reflexes_enabled: false
```

Hello,

I am running into some issues sending repeated rotate goal commands to the create 3 using RCLPY. The code that I am running is pasted below:

```
import rclpy
from rclpy.node import Node
from rclpy.action.client import ActionClient
from rclpy.qos import qos_profile_sensor_data
from rclpy.callback_groups import MutuallyExclusiveCallbackGroup,ReentrantCallbackGroup
from action_msgs.msg._goal_status import GoalStatus

from irobot_create_msgs.action import DriveDistance, Undock,RotateAngle
from irobot_create_msgs.msg import HazardDetectionVector

from pynput.keyboard import KeyCode
from key_commander import KeyCommander
from threading import Lock
from rclpy.executors import MultiThreadedExecutor
from time import sleep,time

# To help with Multithreading
lock = Lock()

class Slash(Node):
    """
    Class to coordinate actions and subscriptions
    """

    def __init__(self, namespace):
        super().__init__('slasher')

        # 2 Seperate Callback Groups for handling the bumper Subscription and Action Clients
        cb_Subscripion = MutuallyExclusiveCallbackGroup()
        #cb_Action = cb_Subscripion
        cb_Action =MutuallyExclusiveCallbackGroup()

        # Subscription to Hazards, the callback function attached only looks for bumper hits
        self.subscription = self.create_subscription(
            HazardDetectionVector, f'/{namespace}/hazard_detection', self.listener_callback, qos_profile_sensor_data,callback_group=cb_Subscripion)

        # Action clients for movements
[... 322 lines cut ...]
Looking at the logs for the create3 for these actions we can see that the rotate goals are received 

`[create3_05F8.motion_control]: Received new rotate_angle goal`

but get stuck on this log message:

`[create3_05F8.motion_control]: rotate_angle goal with angle 3.140000, max_speed 1.900000, per wheel 221`

The Rotate goals that succeed have these following messages:
```
[create3_05F8.motion_control]: Received new rotate_angle goal
[create3_05F8.motion_control]: rotate_angle goal with angle 3.140000, max_speed 1.900000, per wheel 221
[create3_05F8.mobility_monitor]: Received mobility notification: ROBOT_NOTIFY_GYRO_ROTATION_TEST_DONE (56) with arg 0
[create3_05F8.motion_control]: rotate_angle goal succeeded: rotated commanded angle
```

Interestingly, in my testing i found that this issue occurs so far with just multiple rotate goals in a row.  If you have a loop containing a bunch of drive goals they work as expected and all execute for every iteration. I also found that if you rotate once, then perform a drive goal the create3 will drive and rotate for all iterations of the loop as normal.

All of this leads me to believe there may be an issue with the RotateAngle action server on the create3. Somewhere I think things are getting locked up on the create3 for consecutive rotate goals. The goal_uuid_status for the consecutive rotate goal seems to be set properly as executing, but the mobility monitor message never appears in the log and the create3 just sits idle until the goal is cancelled.

Let me know if you need more info from me to try and fix this issue.

I have not tested this issue yet on the new H.1.0 firmware, but will and report back the results.

</div>

Jacob-Friedberg, 2023-03-10:
I have tested this issue out on H.1.0 and the problem persists. Same behavior.

alsora, 2023-03-11:
Closing as duplicate of https://github.com/iRobotEducation/create3_docs/discussions/350

alsora, 2023-03-11:
Hi @Jacob-Friedberg,
We are aware of the problem and we are going to push a fix soon.

As a temporary solution, we recommend you to change RMW from Fast-rtps to cyclonedds.
This should be done both on the robot (via the webserver) as well as on the laptop.

Jacob-Friedberg, 2023-03-11:
Great! I'm glad it's being worked on. However, i would like to mention that the RMW that I'm using on both the create3 and my laptop is cyclonedds already. Did you mean i should swap to fast-rtps?

alsora, 2023-03-11:
Interesting.
We found a bug in fast-rtps that results in the same exact symptoms when sending action or service requests in a loop.
This can happen with all ROS 2 actions or services and is usually triggers after ~10 iterations of the requests loop.
This bug comes from the rmw_fastrtps code and we couldn't reproduce the problem with cyclonedds.

There may be an additional bug specific to the rotate goal action.
We will look into it and keep you posted!
Thank you for reporting this.