iRobotEducation/create3_docs #353: Bluetooth connection failure

fmorshed52, 2023-03-13:
I bought the latest (new) Create 3 iRobot last week.  Set it up (charged, update the firmware, etc.).  But the Bluetooth (blue) light does not come on.

Per instructions, I opened the front, and see that the "switch" is already turned to Bluetooth side.  I rebooted.  Still no blue light.  From my PC I do see that the "Bluetooth & devices" tab says "Bluetooth - Discoverable as "BOOK-somenumber".  But, when I click Add device, and pick  Bluetooth, it cannot find the device.

What is surprising is that the laptop says that it is discoverable, but it cannot connect to it.  What is not surprising, of course, is that the blue light on the iRobot does not come on, hence it cannot connect to it.

Any ideas?

shamlian, 2023-03-13:
I have seen it happen once or twice that the adapter board can get dislodged during shipping. Can you try pulling it out (through the rear of the robot) and then pushing it back in to fully seat the edge connector (like putting in an NES cartridge)?

shamlian, 2023-03-13:
If this doesn't work, what other LEDs are illuminated on the adapter board?

fmorshed52, 2023-03-13:
I pulled the 2 inch wide assembly on right hand side that holds the Bluetooth (Bluetooth symbol, and USB symbol), and snapped it up into place as you said.  No change.  As far as which lights show, only the circle light shows.  Here are two images:
[image]

[image]

shamlian, 2023-03-13:
I can see from that picture that your [adapter board](https://iroboteducation.github.io/create3_docs/hw/adapter/) is not fully seated. Could you pull it out again, and make sure you can see the contacts here? 
[image]

I wonder if a protective plug might have accidentally been left in the connector which is stopping it from seating. When it is fully seated, that "DD" text on the top of the adapter should no longer be visible, like this:
[image]