iRobotEducation/create3_docs #398: Why would the create3 WiFi re-activate when it is configured with Fast-DDS Discovery Server?

royito55, 2023-06-01:
It is my understanding that, for whatever reason, the create3 is disconnected from WiFi when using the Fast-DDS Discovery Server (https://turtlebot.github.io/turtlebot4-user-manual/setup/discovery_server.html)

This leaves only one possible IP for the Discovery Server to work in: the `usb0` interface.

However, after some time, I am able to ping the IP for the create3 without changing any configuration. Why does this happen?

Also, why would iRobot include an option for an IP if you can only use one? Why does it even have to be disconnected from WiFi?

It would be way easier if I can just set the ROS_DISCOVERY_SERVER variable to a different PC in the network. But I can't, because the WiFi stops working with the discovery server. 

Add the weird reconnection to WiFi whenever the base feels like it and there is a really terrible networking problem here. The main appeal of the create3 is to use tons of them, so it is like basic that they should be able to communicate in the same network.

shamlian, 2023-06-23:
I think there is a misunderstanding here; the Create 3 robot can use a Fast-DDS Discovery server while connected to Wi-Fi. Currently, the only way to get the robot to forget its Wi-Fi connection (once connected to a network) is to perform a factory reset. The workaround for now, if you want the robot to stay off a Wi-Fi network, is to either configure the robot through its Soft-AP mode after a factory reset, or else to connect to a temporary Wi-Fi network that can be disabled (for example, a phone hotspot). There is currently an issue (#110) about this missing feature; please 👍 it if you are interested in having it addressed in a future update.

royito55, 2023-06-23:
My bad, I was simply not understanding the configuration.

closed 2023-06-23 by royito55