iRobotEducation/create3_docs #400: Web Server: Address of Fast DDS discovery server does NOT accept special characters

royito55, 2023-06-06:
Hi,

I found an issue while attempting to communicate 2 create3's (with a RPi4) with another PC. 

In order to see both robots, the discovery servers must be run on different IDs. 

This means that the nodes running from the create3 must have the variable `ROS_DISCOVERY_SERVER=;192.168.186.3:11811` (note the preceding `;`). Refer to DDS docs [here](https://fast-dds.docs.eprosima.com/en/latest/fastdds/env_vars/env_vars.html#env-vars-ros-discovery-server) and to my discussion with @EduPonz [here](https://github.com/eProsima/Fast-DDS/discussions/3560#discussioncomment-6077189).

However, when I try to add this in the web server App config, I get the message "Please set a valid ipv4 address and port for the discovery server."
[image]

Can this be changed so special characters are allowed in that field?

Thank you

EduPonz, 2023-06-07:
Thanks @royito55! @alsora could you please point to the correct person to address this?

alsora, 2023-06-07:
I don't have a lot of context on the discovery server, but this seems a fairly straight-forward change.
We have a new firmware that will be released soon and we can make this change there.

EduPonz, 2023-06-07:
Basically, adding `;` as an allowed character there will allow for:
1. Connecting the create3 to several Discovery Servers
2. Connecting the create3 to Discovery Servers with ID other than 0

My guess is that is a simple change on the input validation, but I cannot be sure of course.

alsora, 2023-06-07:
Yes, there's a regex in there and I just added the `;` character.

I'll let you know as soon as the new firmware is publicly available.

PR #420 (2023-07-06): Release G.5.1 / H.2.1
## Changelog
### ROS 2
* Publications
    * Add `/cliff_intensity` topic (fixes #320)
    * Add tf frame for the IMU (called `imu`), to which the `/imu` topic is now connected
* Middleware
    * Allow the use of `;` characters in Fast-DDS Discovery Server configuration (fixes #400)
    * Beta: Allow user to provide XML configuration file for their preferred middleware (fixes #407)
### Core Robot
* Webserver / OS
    * Beta: Allow user full control to edit ntp.conf (fixes #333)
    * Remove erroneous log messages about the robot's lost Wi-Fi connection when it had never be [... cut]
closed 2023-07-06 by shamlian