iRobotEducation/create3_docs #401: USB Network Not Being Polled for NTP

AlexanderDavid, 2023-06-08:
Hi,

I have a Chrony server running on `192.168.186.3`, but I can't see anywhere in the logs that the server is being polled. The only related NTP lines are failing to reach the iRobot pool because the network has no internet access:

```
Jun  8 15:25:29 iRobot-66C91A09E3FD4C11A11F9252BB06CF42 user.notice ntpd: Server 'ntpd' was stopped gracefully.
Jun  8 15:25:34 iRobot-66C91A09E3FD4C11A11F9252BB06CF42 user.notice ntpd: ntpd: bad address '0.irobot.pool.ntp.org'
Jun  8 15:25:39 iRobot-66C91A09E3FD4C11A11F9252BB06CF42 user.notice ntpd: ntpd: bad address '1.irobot.pool.ntp.org'
Jun  8 15:25:44 iRobot-66C91A09E3FD4C11A11F9252BB06CF42 user.notice ntpd: ntpd: bad address '2.irobot.pool.ntp.org'
Jun  8 15:25:49 iRobot-66C91A09E3FD4C11A11F9252BB06CF42 user.notice ntpd: ntpd: bad address '3.irobot.pool.ntp.org'
Jun  8 15:25:55 iRobot-66C91A09E3FD4C11A11F9252BB06CF42 user.notice ntpd: ntpd: bad address '3.irobot.pool.ntp.org'
```

The Create3 base also does not show up when running `sudo chronyc clients`. This server is able to sync time over the non-USB interface correctly, so I'm at a loss as to what to do here. I would expect there to be a similar line for the USB interface. I'm running version G.4.5.

Jayden-F, 2023-06-13:
To address this issue we currently have this band aid fix #327 on the webserver.

- Step 1: Using web server set the create3 time manually using the "Set date and time" Menu to be really far into the past.
- Step 2: Then use the restart ntpd and cross all your fingers and toes; hoping that the create3 accepts the time from the raspberry pi due to large the discrepancy 

This fix would allow user to have more control over the chrony service. However, unfortunately is taking a considerable amount of time to implement: #333

"If only we had root access so we could fix issues ourselves. Without the overhead of building the framework into a webserver"

royito55, 2023-06-27:
@Jayden-F thanks for sharing the band aid fix! I hope there is a better solution soon.

If the ntpd keeps timing out, I'm assuming I didn't set the date and time in the web server far enough in the past?

A lot of these for different addresses: `user.notice ntpd: ntpd: timed out waiting for 185.179.104.7, reach 0x00, next query in 65s`

How do we know when the create3 accepts the time from the pi? Did you do an hour into the past, a day, a month?

Jayden-F, 2023-06-27:
I typically set it back a couple of years. Then check the about tab to see if it has successfully time synced.

shamlian, 2023-07-06:
We have just released G.5.1 / H.2.1 which allows you to directly edit ntp.conf; please feel free to reconfigure the daemon as you please, and let us know when you find a configuration which works in your particular use case. The Create 3 robot does not run `chrony`.

shamlian, 2023-07-11:
In case it helps: Clearpath has an example ntp.conf for the Turtlebot 4 use case; perhaps this can help with yours. https://github.com/turtlebot/turtlebot4/issues/216

AlexanderDavid, 2023-07-19:
@shamlian I finally got a chance to test out the new update and the ntp.conf modification works perfectly. Thanks!

closed 2023-07-19 by AlexanderDavid

shamlian, 2023-07-19:
That's great! Did you end up using the ntp.conf suggested by @roni-krenin ?

AlexanderDavid, 2023-07-19:
I just used my own NTP config. I'm running these on an air gapped network
so I didn't need to try and connect to the irobot pool