iRobotEducation/create3_docs #404: Is there a way to lower the min_speed in motion_control node?

etigafy, 2023-06-14:
Is there a way to lower the min_speed? I want to use cmd_vel topic for a docking procedure and the velocities have to be small to establish accurate docking. The min_speed is set to 0.3 and the parameter is read only.

```
ros2 param describe /motion_control min_speed
Parameter name: min_speed
  Type: double
  Description: Minimum speed in m/s required to move the robot.
  Constraints:
    Read only: true
 ```

alsora, 2023-06-14:
Hi @etigafy the `min_speed` is not a tunable parameter, but rather it is an hardware limitation.
It is exposed as a read-only parameter to allow users to take it into account in their control algorithms.
(indeed our proprietary docking algorithm uses it).

Having said that, the limit is 0.03 m/s, not 0.3 m/s.

Does
```
ros2 param get /motion_control min_speed
```
return 0.3 ?
If yes, there must be a typo in the application and we will fix it in the upcoming release.

etigafy, 2023-06-14:
Thanks for the fast response and the clarification! You're right it was my typo - the 0.03 are correctly displayed when using:
```
ros2 param get /motion_control min_speed
```

alsora, 2023-06-15:
The `0.03 m/s` velocity limit is actually applied per wheel, so you can use that to infer the limit on angular velocities.

Requesting lower velocities currently triggers a saturation limit that scales the velocity up to the minimum.

CraftyCranberry, 2024-04-01:
Hey @alsora I understand that the true max velocity limit is 0.46 m/s. I saw it in the motion_control_node.hpp
Since I am using this for simulation I wanted to increase the limit to 1m/s but even after changing it in this file it still goes to a max of 0.46 when safety override is set to full. Any clue what I'm missing?

alsora, 2024-04-01:
@CraftyCranberry you need to also modify these two lines in the controller configuration: https://github.com/iRobotEducation/create3_sim/blob/main/irobot_create_common/irobot_create_control/config/control.yaml#L60-L61

P.S. if you have further questions about the simulator, please open a dedicated ticket.
I'm closing this ticket as it served its original purpose

closed 2024-04-01 by alsora