iRobotEducation/create3_docs #406: Robot stalls when using python SDK and wheel speeds

TAC0C4T, 2023-06-15:
### Discussed in https://github.com/iRobotEducation/create3_docs/discussions/405

<div type='discussions-op-text'>

<sup>Originally posted by **TAC0C4T** June 14, 2023</sup>
### How are you connecting to your Create 3?

Bluetooth (Python SDK)

### Computer(s) Model(s) and Operating System(s)

Thinkpad P43s Windows 11

### Which version of ROS 2 is installed on your computer?

None

### Which firmware version is installed on your robot?

G.4.3

### Which RMW is your robot running?

I don't know

### Are there multiple Create 3 robots connected to your network?

None

### Is multicast enabled?

None

### What is the Adapter Board's USB/BLE Toggle currently switched to?

None

### Describe your question.

certain values cause a stall when using robot.set_wheel_speeds(left, right). See attached image. Why do these ones cause this behavior?
[image]

Edit: I believe this to most likely be a firmware issue (correct me if I am wrong) as this error persists when the robot is held upside down and there is no resistance on the wheels. This still happens for the green values and the red values do not stall in the same manner.
</div>

lukeopteran, 2023-09-12:
I have also found sometime while turning the robot via /cmd_vel it can turn quickly then it gets locked out, preventing further robot movement. Turning on and off the e-stop (ros service) allows it to move again. Running Galactic + FW5.2, also happened with previous versions

shamlian, 2023-09-12:
Might this have to do with reflexes being enabled, and conflicting with commands sent? What is the status of the `reflexes_enabled` [parameter](https://iroboteducation.github.io/create3_docs/api/reflexes/)?