iRobotEducation/create3_docs #414: NTP does not work H.1.2

royito55, 2023-06-28:
* There is no way to sync the create3 to an NPT server or whatever the default configuration is in there.

* I've seen it work 2 out of 200 times probably, and I don't know the conditions for it to work properly.

* The Restart ntpd Beta Feature does nothing. After clicking it, I get this every single time:

```
Jun 28 11:12:25 iRobot- user.notice ntpd: Server 'ntpd' was stopped gracefully.
Jun 28 11:12:42 iRobot- user.notice ntpd: ntpd: timed out waiting for 162.159.200.123, reach 0x00, next query in 2s
Jun 28 11:12:42 iRobot- user.notice ntpd: ntpd: timed out waiting for 178.32.88.247, reach 0x00, next query in 2s
Jun 28 11:12:42 iRobot- user.notice ntpd: ntpd: timed out waiting for 178.215.228.24, reach 0x00, next query in 2s
Jun 28 11:12:42 iRobot- user.notice ntpd: ntpd: timed out waiting for 90.165.120.190, reach 0x00, next query in 2s
Jun 28 11:13:01 iRobot- user.notice ntpd: ntpd: timed out waiting for 2.59.235.35, reach 0x00, next query in 1s
Jun 28 11:13:01 iRobot- user.notice ntpd: ntpd: timed out waiting for 178.32.88.247, reach 0x00, next query in 2s
Jun 28 11:13:01 iRobot- user.notice ntpd: ntpd: timed out waiting for 89.140.186.3, reach 0x00, next query in 2s
Jun 28 11:13:01 iRobot- user.notice ntpd: ntpd: timed out waiting for 158.227.98.15, reach 0x00, next query in 1s
```

* I saw the fix tried here: https://iroboteducation.github.io/create3_docs/setup/compute-ntp/

It does not work either, because it is configuring the Create3 as the server, but it seems the server never starts correctly.

* I can't disconnect the create3 from WiFi because I'm using CycloneDDS and I haven't found a way to see the topics coming from the create3 with a CycloneDDS config file. If anyone has managed this I'm all ears!

* I can't switch to Fast-DDS to disconnect the create3 because of the many issues involving nav2 (I've successfully ran it in CycloneDDS countless times) described here:
https://github.com/ros2/ros2/issues/1253
https://github.com/ros-planning/navigation2/issues/3033

Hopefully the user will have more "configuration power" and not have the create3 be a black box that prevents it from working unless it is in very simple setups.

royito55, 2023-06-28:
I forgot to mention, I don't think it's a firewall issue either, I tried pinging those addresses that `ntpd` can't find from another machine in the same network and it arrives no problem.

royito55, 2023-06-30:
I found that for this case, the best "solution" is to factory reset the create3

shamlian, 2023-07-06:
Not sure what to say here except that the log snippet you posted does indeed show the NTP server restarting.

You will find that we have just released G.5.1 / H.2.1 which allows direct configuration of ntp.conf; please feel free to reconfigure the daemon as you please, and let us know when you find a configuration which works in your particular use case.

shamlian, 2023-07-19:
Checking in again -- users have had some success with this -- see https://github.com/iRobotEducation/create3_docs/issues/401 . Would be curious to know if this solves your problem as well.

shamlian, 2023-08-21:
Checking in one last time before I close this "cannot reproduce."

closed 2023-09-06 by shamlian