iRobotEducation/create3_docs #432: Waiting for an action server to become available...

mmsaban, 2023-08-10:
I have updated the firmware and am connected to the create3 robot using wifi. The command `ros2 topic echo` displays :

`~/create3_ws$ ros2 topic list
/clock
/diffdrive_controller/cmd_vel_unstamped
/diffdrive_controller/transition_event
/dock_status
/dynamic_joint_states
/imu
/ir_opcode
/joint_state_broadcaster/transition_event
/joint_states
/mouse
/odom
/parameter_events
/performance_metrics
/robot_description
/rosout
/sim_ground_truth_dock_pose
/sim_ground_truth_pose
/standard_dock_description
/tf
/tf_static
`

and I am able to echo the rostopics whichs seems to be working well. However, when sending API commands like `ros2 action send_goal /drive_distance irobot_create_msgs/action/DriveDistance "{distance: 0.5,max_translation_speed: 0.15}"}"
`

i get the following message: 
`Waiting for an action server to become available...
`

How can I ensure the action server to become available? 

Thanks in advance!

mmsaban, 2023-08-10:
Also, while we are in contact. Could you deliberate more onto how to upload a selfmade ros package to the robot? I tried running my own python script (of the subscriber and calling action type). But running it gave the following error: `$ python3 dock_bat.py `

`[WARN] [1691674180.325522731] [battery_monitor]: New publisher discovered on topic '/battery_state', offering incompatible QoS. No messages will be received from it. Last incompatible policy: RELIABILITY`

brianabouchard, 2023-08-10:
Hello @mmsaban, I hope you are well. I recently experienced a similar issue with being able to pull a topic list but not being able to use actions. My problem was a mismatch between the RMW on my computer and the RMW on the robot. In my case, I had the RMW on my computer set to Fast DDS and the robot was on Cyclone. You can find more information on setting your RMW on your robot [here](https://iroboteducation.github.io/create3_docs/webserver/application/) and on your computer [here](https://iroboteducation.github.io/create3_docs/setup/xml-config/).

mmsaban, 2023-08-10:
> > Also, while we are in contact. Could you deliberate more onto how to upload a selfmade ros package to the robot? I tried running my own python script (of the subscriber and calling action type). But running it gave the following error: `$ python3 dock_bat.py `
> > `[WARN] [1691674180.325522731] [battery_monitor]: New publisher discovered on topic '/battery_state', offering incompatible QoS. No messages will be received from it. Last incompatible policy: RELIABILITY`
> 
> It's not possible to directly load a custom package onto the robot itself. However, you can create a package on your computer that interfaces with the robot's existing topics, actions, etc. As for your specific error, would you mind sharing your python script? It looks a bit like you might be setting up a publisher rather than a subscriber.

It is both a subscriber and publisher. I used the publisher as a sanity check to understand when the program triggered some commands. But the script did work in the ros2 simulation environment.