iRobotEducation/create3_docs #446: Could Create 3 work together with Rasp. Pi 5?

knight-arturia, 2023-10-20:
Hi, Dear iRobot team,
I have an issue with the new Raspberry Pi 5, working with Rasp. CAM3 and Create 3. Since I find a connect guidance on Raspberry Pi 4 in Docs. Could Rasp. 5 also connect the adapter board with one USB-C cable, just like the Rasp. 4?
Best Regards

shamlian, 2023-10-23:
You may be the first user of the Create 3 robot who also has access to a Raspberry Pi 5. I don't know any reason why the new board shouldn't have an Ethernet gadget capability on its USB-C port, but I also don't have any data to know, or a device to test.

shamlian, 2023-10-23:
I did some light Internet searching and found https://forums.raspberrypi.com/viewtopic.php?p=2147672&hilit=gadget#p2147672 . Looks like you need to do a little extra work to enable the gadget mode in Bookworm. Perhaps that's the problem? If so, this problem should be reproducible on a Pi 4, as well. What version of the Pi OS are you using?

knight-arturia, 2023-10-23:
> You may be the first user of the Create 3 robot who also has access to a Raspberry Pi 5. I don't know any reason why the new board shouldn't have an Ethernet gadget capability on its USB-C port, but I also don't have any data to know, or a device to test. If you give us some more information about what you've tried to do, we might be able to try to offer suggestions about how to help.

Thanks a lot for your answer.
We are trying to connect Pi 5 to Create 3 and run some object detection algorithms on it.  And send the information back to the server via WiFi.

For the OS, we plan to use Ubuntu Server 22.04, exactly as shown in the Tutorial.

slowrunner, 2024-03-01:
Disclaimer:  I am a Create3 user.  No affiliation with iRobotEd

I have implemented the Pi5 as the compute board for the Create3 successfully.  It is not for the faint of heart, but doable.  Having to run ROS 2 Humble in a Docker container complicates everything, especially having to constantly add something to the docker file, delete the old images to save SDCard space, and then 15-25 minutes to rebuild Docker.

Getting the USB0 setup on Pi5 was not easy, but eventually figured it out, as well as setting up chrony  in PiOS outside docker and the discovery server inside the docker container.  Still not sure I have the super client xml file correct, but all parts (Create3, nodes inside Docker on Pi5, and rviz2 on desktop)  are communicating well.

Still testing the latest firmware H2.6 which allows turning off a lot of the functionality not needed and possibly causing issues trying to run RTABmap and the depthai_ros_driver.  Not encouraging but I’m really on the fringe of my knowledge at this point.

Suggest set up with Pi4 until the Create3 supports Jazzy which will run native over 23.04 on Pi5.

jhubberts, 2024-07-26:
Is there any plan to support Jazzy in the near future?

jomoengineer, 2024-11-15:
I would like to revisit this.  Since the Raspberry Pi 5 has been out for sometime, is it supported with the Create 3 and ROS 2 Jazzy?
Are there any power issues by using a Raspberry Pi 5 vs 4?