iRobotEducation/create3_docs #456: Firmware G5.3 causes slow odomtery output (~8Hz) and jerky movement when driven via cmd_vel, which cannot be recovered from by rolling back to previous firmwares

lukeopteran, 2023-11-14:
Using the latest firmware causes a slow down in the robot function / behavior. We get a jerky response to cmd_vel

When rolling back the firmware the robots still have the issues. I cannot see overly high CPU / memory load on the robot.

Setup:
Firmware: G5.3
DDS: Fastrtps
Pi4 inside on Ubuntu 20.04, RO2 Galactic

On a clean robot running up-to G5.2 we do not see this

lukeopteran, 2023-11-14:
This log is from a robot downgraded from G5.3 to G5.2 and still having the slow performance issues, thanks

[create_log.txt](https://github.com/iRobotEducation/create3_docs/files/13350835/create_log.txt)

lukeopteran, 2023-11-15:
@shamlian The robot had really poor performance under the FastRTPS, even when running it really slowly, we also tried disabling all safety and still had the poor performance. However under cyclone it worked really well, when given a range of cmd_vels (Will stick to Cyclone for now).

Does the robot Ubuntu instance have the ip frag time reduced, as this helped when fastrtps got overwhelmed? e.g. net.ipv4.ipfrag_time = 1

shamlian, 2023-11-15:
Create 3 does not run Ubuntu but I will look into whether or not there is some analogous setting.