iRobotEducation/create3_docs #457: Robot doesn't offer ROS2 topics after some time, requiring reboot to come back

lukeopteran, 2023-11-14:
We frequently have to reboot the robot (best via the UI) to discover the ROS2 topics, they are visible initially but then disappear after an hour or two.

I don't know if it is related to this message in the log files:
**Nov 14 13:52:03 iRobot-A8BDD22042AB4DCBBA85072EAED55E2E user.notice create-platform: 1699969923.059798 [9]        tev: ddsi_udp_conn_write to udp/239.255.0.1:9650 failed with retcode -1**

```
Nov 14 13:51:50 iRobot-A8BDD22042AB4DCBBA85072EAED55E2E daemon.notice network[1472]: [INFO] (src/wifi_control.c:1891) wlan monitor event - CTRL-EVENT-SCAN-STARTED 
Nov 14 13:51:52 iRobot-A8BDD22042AB4DCBBA85072EAED55E2E daemon.notice network[1472]: [INFO] (src/wifi_control.c:1891) wlan monitor event - CTRL-EVENT-SCAN-RESULTS 
Nov 14 13:51:52 iRobot-A8BDD22042AB4DCBBA85072EAED55E2E daemon.notice network[1472]: [INFO] (src/wifi_control.c:1891) wlan monitor event - CTRL-EVENT-NETWORK-NOT-FOUND 
Nov 14 13:51:55 iRobot-A8BDD22042AB4DCBBA85072EAED55E2E user.notice create-platform: 1699969915.058337 [9]        tev: ddsi_udp_conn_write to udp/239.255.0.1:9650 failed with retcode -1
Nov 14 13:51:56 iRobot-A8BDD22042AB4DCBBA85072EAED55E2E daemon.err udhcpc[1874]: sending discover
Nov 14 13:51:57 iRobot-A8BDD22042AB4DCBBA85072EAED55E2E daemon.err udhcpc[1874]: sending discover
Nov 14 13:51:57 iRobot-A8BDD22042AB4DCBBA85072EAED55E2E daemon.notice network[1472]: [INFO] (src/wifi_control.c:1891) wlan monitor event - CTRL-EVENT-SCAN-STARTED 
Nov 14 13:51:58 iRobot-A8BDD22042AB4DCBBA85072EAED55E2E daemon.err udhcpc[1874]: sending discover
Nov 14 13:52:00 iRobot-A8BDD22042AB4DCBBA85072EAED55E2E daemon.notice network[1472]: [INFO] (src/wifi_control.c:1891) wlan monitor event - CTRL-EVENT-SCAN-RESULTS 
Nov 14 13:52:00 iRobot-A8BDD22042AB4DCBBA85072EAED55E2E daemon.notice network[1472]: [INFO] (src/wifi_control.c:1891) wlan monitor event - CTRL-EVENT-NETWORK-NOT-FOUND 
Nov 14 13:52:03 iRobot-A8BDD22042AB4DCBBA85072EAED55E2E user.notice create-platform: 1699969923.059798 [9]        tev: ddsi_udp_conn_write to udp/239.255.0.1:9650 failed with retcode -1
Nov 14 13:52:05 iRobot-A8BDD22042AB4DCBBA85072EAED55E2E daemon.notice network[1472]: [INFO] (src/wifi_control.c:1891) wlan monitor event - CTRL-EVENT-SCAN-STARTED 
Nov 14 13:52:07 iRobot-A8BDD22042AB4DCBBA85072EAED55E2E daemon.notice network[1472]: [INFO] (src/wifi_control.c:1891) wlan monitor event - CTRL-EVENT-SCAN-RESULTS 
Nov 14 13:52:07 iRobot-A8BDD22042AB4DCBBA85072EAED55E2E daemon.notice network[1472]: [INFO] (src/wifi_control.c:1891) wlan monitor event - CTRL-EVENT-NETWORK-NOT-FOUND 
Nov 14 13:52:10 iRobot-A8BDD22042AB4DCBBA85072EAED55E2E daemon.notice network[1472]: [INFO] (src/gateway_pinger.c:121) wifi is disconnected
Nov 14 13:52:10 iRobot-A8BDD22042AB4DCBBA85072EAED55E2E daemon.notice network[1472]: [INFO] (src/gateway_pinger.c:219) no gateway to ping
Nov 14 13:52:11 iRobot-A8BDD22042AB4DCBBA85072EAED55E2E user.notice create-platform: 1699969931.059229 [9]        tev: ddsi_udp_conn_write to udp/239.255.0.1:9650 failed with retcode -1
Nov 14 13:52:11 iRobot-A8BDD22042AB4DCBBA85072EAED55E2E user.notice create-platform: [INFO] [1699969931.208929480] [system_health]: CPU usage: max 56 [%] mean 52 [%] RAM usage: 29/59 [MB]
```

### Robot setup
Firmware: G5.2
Host: Pi4 inside Ubuntu 20.04 on usb0 ROS galactic, running chrony as a ntp server

brianabouchard, 2023-11-14:
Hello! I haven't seen this behavior before, but I will try to replicate it. Can you tell me a little more about your RMW setup? Are you using Fast DDS or Cyclone and do you have a custom XML configuration on the Pi?

shamlian, 2023-11-14:
I believe that error happens when Cyclone DDS has trouble due to a network change, so it is quite possible that log message is an indicator. Is something changing on your network? The log seems to indicate that your robot is losing a network connection. In my experience, Cyclone DDS is quite sensitive to changes in network settings and does not recover well when an interface goes down.

brianabouchard, 2023-11-16:
Hi @lukeopteran! I've run some tests and re-examined your logs. I was not able to replicate what you are seeing by just letting it sit idle for several hours. However, I was able to replicate the error you're seeing in the logs by removing the usb0 IP assignment from the netplan yaml on the Pi, applying the new netplan, restarting the Pi, and adding back and re-applying the usb0 settings. I ran this check to see if a disruption in that interface would cause an issue, and it did. My suspicion is that you may have a faulty USB-C connection. Are you able to test with a new cable and then upload a full copy of your logs? I'd love to capture the logs particularly right before you start getting the failed with retcode -1 errors.

shamlian, 2023-11-17:
Just to add to this -- it would be extremely helpful if you had logs from the robot when this problem begins to happen, so we can try to reproduce the problem here. You've provided a snippet, but having the context and the event would be _extremely_ helpful. If you are uncomfortable uploading the logs here, you can email them to education@irobot.com and they will make their way to the Create 3 team.

lukeopteran, 2023-11-23:
We have multiple robots with this setup, with different usbc leads, I don't think we have a physical poor connection. It also happens after a while when the robots aren't moving.

Here is a log file from just now, a reboot fixed it:
[messages no ros2 topics until reboot.txt](https://github.com/iRobotEducation/create3_docs/files/13449292/messages.no.ros2.topics.until.reboot.txt)

brianabouchard, 2023-11-26:
Thanks for the update. It looks like these logs don't capture the point at which the connection stopped working, which may be hard to do with your setup since it is hard to predict exactly when it goes down. 

I've run a series of tests over the last 48 hours and am still unable to replicate the issue you are seeing. The Create 3 robot provided a topic list after 1 hour, 2 hours, and a full 24 hours with no reboot. Based on the information you've provided, there are two differences in our set ups. First, I factory reset my Create 3 robot to stop it from attempting to join a Wi-Fi network. The second is that I am not using any custom xml profiles for either the Create or the Pi. I'm going to run some tests with your settings now, but while I do, it would be helpful if you could attempt to replicate my set up so we can try to further isolate the root of the issue.

brianabouchard, 2023-11-27:
Hi again, I ran another overnight test replicating all of the settings you have provided, and I'm still able to receive a topic list after 12-18 hours of inactivity. Since I was only able to replicate the errors you are seeing in the logs by disrupting or altering the usb0 interface, I still suspect this is the cause. If you have tried different cables, then my next thought would be it is some sort of OS issue that is putting that interface to sleep on the Pi after a certain period of time. Can you please provide additional details on how you've configured your Ubuntu image on your Pi?

lukeopteran, 2024-04-10:
Had some more network over USB comms failures, I got a log with the failure and after when reboot and working again.

Running latest Humble H2.6FW and cyclone custom RMW config

[after full restart working again.txt](https://github.com/iRobotEducation/create3_docs/files/14933099/after.full.restart.working.again.txt)
[usb_comms_fail.txt](https://github.com/iRobotEducation/create3_docs/files/14933102/usb_comms_fail.txt)

shamlian, 2024-04-10:
I'm wondering if there is a problem with the physical link. How well is your cable strain-relieved? Could it be bouncing around? Is it in good shape? Also, on the Pi, are there any interesting messages in `dmesg` or `/var/log/messages` when this happens?

lukeopteran, 2024-04-10:
The robot wasn't moving when the fault occurred, will check the logging

lukeopteran, 2024-04-15:
It has done it again, the robot is sat completely still, but has been on (charging) for a good while (>5 hours). I did software restart of the Pi at one point a while ago.

I get no ROS2 messages, but I can get to the Create3 web UI etc fine through the same network route (Pi > usb0)...

[messages (5).txt](https://github.com/iRobotEducation/create3_docs/files/14979529/messages.5.txt)

Here are any messages in /var/log relating to the usb0 connection

```
boot.log:[   18.185007] cloud-init[863]: ci-info: |  usb0  |  True |        192.168.186.1         | 255.255.255.0 | global | e2:d5:92:d4:cf:90 |
boot.log:[   18.185183] cloud-init[863]: ci-info: |  usb0  |  True | fe80::e0d5:92ff:fed4:cf90/64 |       .       |  link  | e2:d5:92:d4:cf:90 |
boot.log:[   18.186482] cloud-init[863]: ci-info: |   0   | 192.168.186.0 | 0.0.0.0 | 255.255.255.0 |    usb0   |   U   |
boot.log:[   18.187622] cloud-init[863]: ci-info: |   0   |  fe80::/64  |    ::   |    usb0   |   U   |
boot.log:[   18.187802] cloud-init[863]: ci-info: |   1   |    local    |    ::   |    usb0   |   U   |
boot.log:[   18.188122] cloud-init[863]: ci-info: |   2   |  multicast  |    ::   |    usb0   |   U   |
kern.log:Nov 21 21:10:38 irobot37 kernel: [   23.668504] IPv6: ADDRCONF(NETDEV_CHANGE): usb0: link becomes ready
kern.log:Nov 21 21:10:39 irobot37 kernel: [   23.903330] IPv6: ADDRCONF(NETDEV_CHANGE): usb0: link becomes ready
boot.log.5:[   18.304626] cloud-init[861]: ci-info: |  usb0  | False |     .     |     .     |   .   | 96:ab:29:04:c6:6e |
boot.log.5:[   17.899344] cloud-init[868]: ci-info: |  usb0  | False |     .     |     .     |   .   | be:30:19:4b:31:f4 |
boot.log.5:[   18.233844] cloud-init[871]: ci-info: |  usb0  | False |     .     |     .     |   .   | 22:5d:43:27:8d:f0 |
syslog:Apr 15 12:25:11 irobot37 avahi-daemon[891]: Leaving mDNS multicast group on interface usb0.IPv6 with address fe80::9cf9:aeff:fe42:f1e9.
syslog:Apr 15 12:25:11 irobot37 avahi-daemon[891]: Leaving mDNS multicast group on interface usb0.IPv4 with address 192.168.186.1.
syslog:Nov 21 21:10:34 irobot37 systemd-udevd[485]: usb0: Could not generate persistent MAC: No data available
syslog:Nov 21 21:10:34 irobot37 systemd-networkd[856]: usb0: IPv6 successfully enabled
syslog:Nov 21 21:10:34 irobot37 systemd-networkd[856]: usb0: Link UP
syslog:Nov 21 21:10:34 irobot37 systemd-networkd[856]: usb0: Gained carrier
syslog:Nov 21 21:10:34 irobot37 systemd-networkd[856]: usb0: Gained IPv6LL
syslog:Nov 21 21:10:34 irobot37 cloud-init[863]: ci-info: |  usb0  |  True |        192.168.186.1         | 255.255.255.0 | global | e2:d5:92:d4:cf:90 |
syslog:Nov 21 21:10:34 irobot37 cloud-init[863]: ci-info: |  usb0  |  True | fe80::e0d5:92ff:fed4:cf90/64 |       .       |  link  | e2:d5:92:d4:cf:90 |
syslog:Nov 21 21:10:34 irobot37 cloud-init[863]: ci-info: |   0   | 192.168.186.0 | 0.0.0.0 | 255.255.255.0 |    usb0   |   U   |
syslog:Nov 21 21:10:34 irobot37 cloud-init[863]: ci-info: |   0   |  fe80::/64  |    ::   |    usb0   |   U   |
syslog:Nov 21 21:10:34 irobot37 cloud-init[863]: ci-info: |   1   |    local    |    ::   |    usb0   |   U   |
syslog:Nov 21 21:10:34 irobot37 cloud-init[863]: ci-info: |   2   |  multicast  |    ::   |    usb0   |   U   |
syslog:Nov 21 21:10:35 irobot37 avahi-daemon[891]: Joining mDNS multicast group on interface usb0.IPv6 with address fe80::e0d5:92ff:fed4:cf90.
syslog:Nov 21 21:10:35 irobot37 avahi-daemon[891]: New relevant interface usb0.IPv6 for mDNS.
syslog:Nov 21 21:10:35 irobot37 avahi-daemon[891]: Joining mDNS multicast group on interface usb0.IPv4 with address 192.168.186.1.
syslog:Nov 21 21:10:35 irobot37 avahi-daemon[891]: New relevant interface usb0.IPv4 for mDNS.
syslog:Nov 21 21:10:35 irobot37 avahi-daemon[891]: Registering new address record for fe80::e0d5:92ff:fed4:cf90 on usb0.*.
syslog:Nov 21 21:10:35 irobot37 avahi-daemon[891]: Registering new address record for 192.168.186.1 on usb0.IPv4.
syslog:Nov 21 21:10:36 irobot37 NetworkManager[893]: <info>  [1700601036.5290] device (usb0): carrier: link connected
syslog:Nov 21 21:10:36 irobot37 NetworkManager[893]: <info>  [1700601036.5302] manager: (usb0): new Ethernet device (/org/freedesktop/NetworkManager/Devices/3)
Binary file cloud-init.log matches
syslog.1:Nov 21 21:10:34 irobot37 systemd-udevd[465]: usb0: Could not generate persistent MAC: No data available
syslog.1:Nov 21 21:10:34 irobot37 systemd-udevd[489]: usb0: Could not generate persistent MAC: No data available
syslog.1:Nov 21 21:10:34 irobot37 systemd-networkd[856]: usb0: IPv6 successfully enabled
syslog.1:Nov 21 21:10:34 irobot37 systemd-networkd[856]: usb0: Link UP
syslog.1:Nov 21 21:10:34 irobot37 systemd-networkd[856]: usb0: Gained carrier
syslog.1:Nov 21 21:10:34 irobot37 systemd-networkd[856]: usb0: Gained IPv6LL
syslog.1:Nov 21 21:10:34 irobot37 cloud-init[863]: ci-info: |  usb0  |  True |        192.168.186.1         | 255.255.255.0 | global | e6:06:4d:54:8c:13 |
syslog.1:Nov 21 21:10:34 irobot37 cloud-init[863]: ci-info: |  usb0  |  True | fe80::e406:4dff:fe54:8c13/64 |       .       |  link  | e6:06:4d:54:8c:13 |
syslog.1:Nov 21 21:10:34 irobot37 cloud-init[863]: ci-info: |   0   | 192.168.186.0 | 0.0.0.0 | 255.255.255.0 |    usb0   |   U   |
syslog.1:Nov 21 21:10:34 irobot37 cloud-init[863]: ci-info: |   0   |  fe80::/64  |    ::   |    usb0   |   U   |
syslog.1:Nov 21 21:10:34 irobot37 cloud-init[863]: ci-info: |   1   |    local    |    ::   |    usb0   |   U   |
[... cut ...]
auth.log:Apr 11 09:27:04 irobot37 sudo:   ubuntu : TTY=pts/0 ; PWD=/var/log ; USER=root ; COMMAND=/usr/bin/grep -r usb0
auth.log:Apr 15 14:17:50 irobot37 sudo:   ubuntu : TTY=pts/1 ; PWD=/var/log ; USER=root ; COMMAND=/usr/bin/grep -r usb0
boot.log.6:[   17.468695] cloud-init[860]: ci-info: |  usb0  | False |     .     |     .     |   .   | 06:37:e3:aa:a4:19 |
kern.log.1:Nov 21 21:10:38 irobot37 kernel: [   23.933870] IPv6: ADDRCONF(NETDEV_CHANGE): usb0: link becomes ready
kern.log.1:Nov 21 21:10:38 irobot37 kernel: [   24.029286] IPv6: ADDRCONF(NETDEV_CHANGE): usb0: link becomes ready

```

brianabouchard, 2024-04-16:
My experience has generally been that any restart of the Pi can disrupt the ROS 2 connection with the Create 3. I think it's a weird middleware/ROS 2 issue. Whenever I restart the Pi, I also reboot the Create 3 after doing so.

lukeopteran, 2024-04-23:
Hi @brianabouchard, thanks have just found the same, restarting the Pi reliably generates the issue.

Can the network stack on the robot be automatically restarted if it looses usb0 comms, it's always the same error (tev: ddsi_udp_conn_write to udp/239.255.0.1:9650 failed with retcode -1)? As the boot cycle of the robot is very long and its quite easy to forget to do this...

brianabouchard, 2024-04-23:
I can't speak for the iRobot team and whether or not that would be possible. But as a workaround, it should be possible to setup a service that runs on boot and executes either `curl -X POST http://192.168.186.2/api/reboot` or `curl -X POST http://192.168.186.2/api/restart-app`. I haven't tested the latter for this issue, but in theory, restarting the application rather than fully rebooting the robot may resolve the issue and would be faster.

royito55, 2024-09-10:
Hi @lukeopteran, the reason you don't see the topics all of the time is due to discovery issues and the inability of the Create 3 to parse these in time in a DDS network. 

We have made a video that addresses discovery and memory issues using Zenoh. 

I hope it helps!

https://www.youtube.com/watch?v=xmK2I0D5sas