iRobotEducation/create3_docs #528: Bad links on Install ROS 2 Humble with Create 3 Messages on an Ubuntu 22.04 Machine

jomoengineer, 2024-01-24:
There are links on the page for "Install ROS 2 Humble with Create 3 Messages on an Ubuntu 22.04 Machine" that are showing a 404 error.
Page
[https://github.com/iRobotEducation/create3_docs/blob/main/docs/setup/ubuntu2204.md](url)

Ex:
More detail on RMW can be found [here](https://github.com/iRobotEducation/create3_docs/blob/main/docs/xml-config)

If this does not work, please refer to [ROS 2 Network Configuration](https://github.com/iRobotEducation/create3_docs/blob/main/docs/xml-config) for further configuration ideas. A full Create® 3 API description can be found [here](https://github.com/iRobotEducation/create3_docs/blob/main/api/ros2).

shamlian, 2024-01-24:
On it!

shamlian, 2024-01-24:
The links are fine -- the problem is that you are accessing the _source_ of the docs site instead of the docs site itself. Try https://iroboteducation.github.io/create3_docs/setup/ubuntu2204/

jomoengineer, 2024-01-24:
Ah, got it. thanks

closed 2024-01-24 by shamlian