iRobotEducation/create3_docs #533: Bumper cannot be disabled

halimimaha, 2024-01-31:
Robot Model
Turtlebot4 Standard

ROS distro
Humble

Networking Configuration
Simple Discovery

OS
Ubuntu 22.04

Built from source or installed?
Installed

Actual behaviour
I followed this tutorial: https://iroboteducation.github.io/create3_docs/api/reflexes/ but when I send the command to disable the bumber it doesn't work, the bumber is still activate, any suggestions ?

To Reproduce
i sent this command :

ros2 param set /robot1/motion_control reflexes.REFLEX_BUMP false

i try also to add "reflexes.REFLEX_BUMP: false" in the end of this script in web server of create 3 : 

motion_control:
  ros__parameters:
    # safety_override options are 
    # "none" - standard safety profile, robot cannot backup more than an inch because of lack of cliff protection in rear, max speed 0.306m/s
    # "backup_only" - allow backup without cliff safety, but keep cliff safety forward and max speed at 0.306m/s
    # "full" - no cliff safety, robot will ignore cliffs and set max speed to 0.46m/s
    safety_override: "none"
    reflexes.REFLEX_BUMP: false

alsora, 2024-01-31:
Hi,
Which version of the create 3 firmware are you using?

What's the output of this command? 
```
ros2 param set /robot1/motion_control reflexes.REFLEX_BUMP false
```

Can you confirm that the parameter is disabled by running
```
ros2 param get /robot1/motion_control reflexes.REFLEX_BUMP
```

Can you try to disable all reflexes 
```
ros2 param set /robot1/motion_control reflexes_enabled false
```

Lastly, your yaml syntax in the webserver seems wrong. 
You are passing the argument as a string rather than a boolean. It should be:

```
reflexes.REFLEX_BUMP: False
```

halimimaha, 2024-01-31:
Hello, thans for ur reply !

the output of this command : 

ros2 param set /raphaelo/motion_control reflexes_enabled false

is :  
Wait for service timed out 

i tried 1000 times and every time i have the same output :/

halimimaha, 2024-01-31:
and also the output of this comad  : 
ros2 param get /raphaelo/motion_control reflexes.REFLEX_BUMP

is :
Boolean value is: True