iRobotEducation/create3_docs #547: "Simply restart robot" instruction in ntp server config documentation unclear

slowrunner, 2024-02-17:
> [If this happens, simply restart the robot (not just the application) via the webserver over the network connection.](https://iroboteducation.github.io/create3_docs/setup/compute-ntp/#step-by-step)

(Last line of page)

I do not see a "Restart Robot" button or menu item on any of the pages of the webserver application (H.2.4 F/W).

How do I restart the robot via the webserver?

alsora, 2024-02-17:
> the docs say that option is only available when using the Create3 WiFi Access Point:

This is correct. We should modify the ntp page to be more clear.

> Tried using button1 10s to standby mode while on dock
This is expected, as the dock contacts are what wakes up the robot after you turn it off.

slowrunner, 2024-02-17:
> We should modify the ntp page to be more clear.

Love it.

And suggest modify the WebServer Application "Reboot Robot" entry - with something like:
```
If Reboot Robot menu option does not appear (WebServer Application over USB) reboot robot:
- With Create3 not docked, press and hold the center power button for 10 seconds (will hear shutdown chime)
- After light ring goes dark, set Create3 on dock (green dock LED lights)  
```