iRobotEducation/create3_docs #558: H2.5 FW - full safety override ignores /cmd_vel topic

slowrunner, 2024-02-29:
# H2.5 firmware with "full" safety override ignores /cmd_vel topic

Create3 responds to /rotate_angle for example, but not /cmd_vel
- Verified Create3 is publishing
```
$ ros2 topic list
/battery_state
/cliff_intensity
/cmd_audio
/cmd_lightring
/cmd_vel
/dock_status
/interface_buttons
/ir_intensity
/ir_opcode
/joy
/joy/set_feedback
/kidnap_status
/mobility_monitor/transition_event
/odom
/parameter_events
/robot_state/transition_event
/rosout
/slip_status
/static_transform/transition_event
/stop_status
/tf_static
pi@WaLiPi5:~/wali_pi5/c3ws $ 

``` 
- and accepting service and action goals:
```
pi@WaLiPi5:~/wali_pi5/c3ws $ cmds/pub_180.sh 

*** PUB 180
ros2 action send_goal /rotate_angle irobot_create_msgs/action/RotateAngle "{angle: 3.14159,max_rotation_speed: 0.5}"
Waiting for an action server to become available...
Sending goal:
     angle: 3.14159
max_rotation_speed: 0.5

Goal accepted with ID: 82a6435d9e194df687b3971e31a48972

Result:
    pose:
  header:
    stamp:
      sec: 1709222547
      nanosec: 73896258
    frame_id: odom
  pose:
    position:
      x: 0.017357299104332924
      y: -0.0005571765941567719
      z: 9.282828978030011e-05
    orientation:
      x: 0.0014758383014984479
      y: -0.0002464390300625141
      z: 0.9999902545638742
      w: 0.004153546242726723

Goal finished with status: SUCCEEDED

```

- Verified /cmd_vel topic is being published by USB0 connected system:
```
pi@WaLiPi5:~/wali_pi5/c3ws $ ros2 topic echo --once /cmd_vel
linear:
  x: -0.0
  y: 0.0
  z: 0.0
angular:
  x: 0.0
  y: 0.0
  z: -0.4999999701976776
---
pi@WaLiPi5:~/wali_pi5/c3ws $ ros2 topic echo --once /cmd_vel
linear:
  x: 0.14999999105930328
  y: 0.0
  z: 0.0
angular:
  x: 0.0
  y: 0.0
  z: -0.0
---
pi@WaLiPi5:~/wali_pi5/c3ws $ 

```

- Log Output showing /rotate_angle success and status during period /cmd_vel topics should have caused motion:
```
Feb 29 16:02:41 Create3-WaLi user.notice create-platform: [WARN] [1709222561.223861217] [motion_control]: rotate_angle goal succeeded: rotated commanded angle
Feb 29 16:03:25 Create3-WaLi user.notice create-platform: [INFO] [1709222605.433915975] [system_health]: CPU usage: max 46 [%] mean 35 [%] RAM usage: 33/59 [MB]
Feb 29 16:04:25 Create3-WaLi user.notice create-platform: [INFO] [1709222665.406335984] [system_health]: CPU usage: max 45 [%] mean 35 [%] RAM usage: 33/59 [MB]
Feb 29 16:05:25 Create3-WaLi user.notice create-platform: [INFO] [1709222725.591542294] [system_health]: CPU usage: max 54 [%] mean 41 [%] RAM usage: 33/59 [MB]
Feb 29 16:06:25 Create3-WaLi user.notice create-platform: [INFO] [1709222785.594932773] [system_health]: CPU usage: max 81 [%] mean 46 [%] RAM usage: 33/59 [MB]
Feb 29 16:07:25 Create3-WaLi user.notice create-platform: [INFO] [1709222845.415104429] [system_health]: CPU usage: max 47 [%] mean 38 [%] RAM usage: 33/59 [MB]
Feb 29 16:08:25 Create3-WaLi user.notice create-platform: [INFO] [1709222905.408061477] [system_health]: CPU usage: max 50 [%] mean 37 [%] RAM usage: 33/59 [MB]
Feb 29 16:08:29 Create3-WaLi user.notice ntpd: Script called in periodic mode
Feb 29 16:09:25 Create3-WaLi user.notice create-platform: [INFO] [1709222965.407555417] [system_health]: CPU usage: max 54 [%] mean 40 [%] RAM usage: 33/59 [MB]
Feb 29 16:10:25 Create3-WaLi user.notice create-platform: [INFO] [1709223025.409140445] [system_health]: CPU usage: max 56 [%] mean 49 [%] RAM usage: 33/59 [MB]
Feb 29 16:11:25 Create3-WaLi user.notice create-platform: [INFO] [1709223085.411026369] [system_health]: CPU usage: max 69 [%] mean 53 [%] RAM usage: 32/59 [MB]
Feb 29 16:12:25 Create3-WaLi user.notice create-platform: [INFO] [1709223145.413557282] [system_health]: CPU usage: max 51 [%] mean 41 [%] RAM usage: 32/59 [MB]
```

- Issue only occurs when safety_override is set to "full" (only change from proper behavior file with "backup_only" setting):
```
motion_control:
  ros__parameters:
    # safety_override options are 
    # "none" - standard safety profile, robot cannot backup more than an inch because of lack of cliff protection in rear, max speed 0.306m/s
    # "backup_only" - allow backup without cliff safety, but keep cliff safety forward and max speed at 0.306m/s
    # "full" - no cliff safety, robot will ignore cliffs and set max speed to 0.46m/s
    safety_override: "full"

robot_state:
  ros__parameters:
    # publish_hazard_msgs: default true, set false to noticeably reduce CPU and RAM useage
    publish_hazard_msgs: false
    # publish_odom_tfs: default true, if not needed will noticeably reduce CPU and RAM use
    publish_odom_tfs: false 
    # raw_kinematics_min_pub_period_ms: default 25 , -1 disables /imu, /mouse, /wheel_status, /wheel_ticks, /wheel_vels (-1 recommended)
    raw_kinematics_min_pub_period_ms: -1
    # hazards_pub_fixed_period_ms: default -1 disables fixed period hazard publication, pubs only when change noted, 1: publish as soon as possible, 2+ period for constant publishing 

```

[cmd_vel_test_with_safety_override_full_and_backup_only.txt](https://github.com/iRobotEducation/create3_docs/files/14450832/cmd_vel_test_with_safety_override_full_and_backup_only.txt)

shamlian, 2024-02-29:
Oops. We see the source of the bug, will patch soon.

shamlian, 2024-02-29:
For now, if you set the parameter to "full" at run time, you can work around this bug (that is, `ros2 param set /motion_control safety_override "full"`)

shamlian, 2024-02-29:
Hi @slowrunner , thanks for the report. Please open a new bug report if you find anything else!

PR #559 (2024-02-29): Release H.2.6
Bugfix release!

Robot now sets `max_speed` parameter properly when `safety_override = full` is set in application configuration ( closes #558 )
closed 2024-02-29 by shamlian