iRobotEducation/create3_docs #562: Create3 docs show bot killer echo /tf, should show tf2_echo with repetition rate

slowrunner, 2024-03-02:
# See final comment in discussion thread for suggestion

### Discussed in https://github.com/iRobotEducation/create3_docs/discussions/496

<div type='discussions-op-text'>

<sup>Originally posted by **atinfinity** December 22, 2023</sup>
### How are you connecting to your Create 3?

USB Ethernet (ROS 2)

### Which version of ROS 2 is installed on your computer?

Humble

### Which firmware version is installed on your robot?

H.2.3

### Which RMW is your robot running?

FastRTPS

### Are there multiple Create 3 robots connected to your network?

None

### Is multicast enabled?

None

### What is the Adapter Board's USB/BLE Toggle currently switched to?

None

### Describe your question.

I encountered this problem(<https://github.com/turtlebot/turtlebot4/issues/152>).
And, I found the following description in this ticket .

> The issue is being tracked [here](https://github.com/iRobotEducation/create3_docs/issues/381).

<https://github.com/iRobotEducation/create3_docs/issues/381> is reported in Create 3. I checked that CPU usage of Create 3 is very high.
But, this ticket has no updates. Is this a problem of Create 3's firmware?
I think iRobot should clarify this issue.</div>

alsora, 2024-03-02:
I think that this recommendation is not correct.
Using `tf2_echo` requires the create 3 to do more work than just doing `ros2 topic echo /tf`.

I saw your data in the other thread, but correlation is not causation.
I provided details here about why `tf2_echo` shouldn't be used as an alternative to `/tf` to reduce CPU usage on the robot: https://github.com/iRobotEducation/create3_docs/discussions/496#discussioncomment-8653521

Moreover, the numeric arguments that you add at the end
```
ros2 run tf2_ros tf2_echo odom base_link 0.1
```

is the frequency at which your laptop will check if the tree is valid and the transformation is available.
It won't have any impact on the work done by the robot.

The `/tf` topic has nothing special.
Subscribing to it won't cause more load to the robot than subscribing to any other topic with the same publish frequency.

slowrunner, 2024-03-03:
Understand better, thanks.

closed 2024-03-03 by slowrunner