iRobotEducation/create3_docs #570: iRobot Create w/ Jetson Orin - ros2 topic list shows no create topics

catglossop, 2024-03-12:
Hello! I've been having issues with setting up a new create 3. I've followed the create [setup guide](https://iroboteducation.github.io/create3_docs/setup/ubuntu2004/) and have the following configuration: 

**Compute:** Jetson orin nx (j401 from seeed studio)
**Network configuration:** Ethernet over USB (toggle set to USB) 
**Jetpack version**: 5.1
**Ubuntu**: 20.04
**Ros Version:** Galactic
**Firmware version**: G.5.4
**RMW**: rmw_cyclonedds_cpp

My current XML file for cyclonedds looks like as follows:

<CycloneDDS>
   <Domain>
     <General>
        <NetworkInterfaceAddress>l4tbr0</NetworkInterfaceAddress>
    </General>
   </Domain>
</CycloneDDS>

I've also tried with ros2 humble from source with a humble firmware version and using the galactic docker container recommended by the guide. I've followed the steps to set the IP for ethernet over USB and the `l4tbr0` interface is up and can be pinged. The appropriate `irobot-create-msgs` have been installed. No matter what I've tried, `ros2 topic list` only shows `/parameter_events` and `/rosout`. Any suggestions on debugging this? I'm happy to provide more specifics on what I've tried. 

Thanks in advance!

brianabouchard, 2024-03-12:
Hello! My experience has been that your first setup (Galactic native install) will be the most successful with the device model you have and the jetpack version. A few clarifying questions, if I may:

- When you say you can ping the l4tbr0 interface, you are pinging 192.168.186.2, correct?
- Have you specified a domain ID on either the Create 3 or the Orin? 
- For the custom XML file, are you using that on the Orin or are you also trying it on the Create 3? P.S. **Don't** use that XML on the Create 3, if you aren't already. 

I've been working on a troubleshooting script to help specifically with cross-checking ROS 2 settings. To be transparent, I have not tested it on the Orin yet only Pis and VMs, but in theory, it should work on the Orin. If you want to try it, you can download it here: https://github.com/iRobotEducation/create3_docs/blob/brianabouchard/revised-question-template/.github/DISCUSSION_TEMPLATE/troubleshooting.py

Also, I'm guessing you've already tried this, but I would definitely reboot the Create 3 application while the Orin is still connected. You can do this through the webserver.

catglossop, 2024-03-12:
Hi Briana! I had actually only ever rebooted the robot through the webserver, I didn't realize restarting the application would be something not contained in rebooting but that actually fixed it! I can see all the topics now and messages are coming through. Thanks!

closed 2024-03-12 by catglossop