iRobotEducation/create3_docs #573: ROS2 Action Server not working in Humble

AndrewGerstenslager, 2024-03-16:
I think there should be an additional note somewhere in the documentation if there is an incompatibility with a computer running ROS2 Humble and the Create3 in terms of action servers and possibly services (not tested). Here is the behavior as follows:

I ran the /turn_angle action on my PC running Humble. Here is the output:
```
andrew@andrewdesktop:~$ ros2 action send_goal /rotate_angle irobot_create_msgs/action/RotateAngle "{angle: 1.57,max_rotation_speed: 0.5}"
Waiting for an action server to become available...
Sending goal:
     angle: 1.57
max_rotation_speed: 0.5

2024-03-16 15:03:23.911 [RTPS_READER_HISTORY Error] Change payload size of '32' bytes is larger than the history payload size of '19' bytes and cannot be resized. -> Function can_change_be_added_nts
```

When I then installed galactic and ran the same command the robot worked perfectly. I'm not sure if there is a command to make Humble compatible, but I did not see any references to that. If there is an actual issue with action servers with Humble then a disclaimer would have saved me a lot of time.

brianabouchard, 2024-03-17:
Hello! To the best of my knowledge, ROS 2 systems are not intended to be compatible across releases. If you wanted to use the Create 3 on Humble, you can install the Humble firmware on the Create 3. Here's a link to the latest versions of both Galactic and Humble: https://iroboteducation.github.io/create3_docs/releases/overview/

shamlian, 2024-03-18:
@brianabouchard is correct. I am going to close this "not planned." Please see this ROS answer for more details: https://answers.ros.org/question/341372/can-nodes-from-different-ros-2-distributions-communicate-compatibly/ .

closed 2024-03-18 by shamlian

AndrewGerstenslager, 2024-03-18:
@brianabouchard @shamlian thanks for the quick responses. My issues were less on the need to install another ROS2 version (which I had done) but moreso a suggestion to add a note about cross version compatibility on your documentation. Additionally if there is a way to figure out the version of ROS2 on the robot I'd suggest making that more clear as I couldn't find an obvious way to do that looking at the docs. Thanks for your help!

shamlian, 2024-03-18:
The page @brianabouchard linked you to says, "Create® 3 releases take the version schema <ROS 2 letter>.<major number>.<minor number>. The letter corresponds with the [ROS 2](https://iroboteducation.github.io/create3_docs/releases/overview/#fn:1) Release. For example, G corresponds with Galactic, and H corresponds with Humble. See [ROS 2 Releases](https://docs.ros.org/en/rolling/Releases.html)." It's also available on the ["update"](https://iroboteducation.github.io/create3_docs/webserver/update/) page.

Would be happy to add information to other pages if you have suggestions, and glad you're all set.

shamlian, 2024-03-18:
Or -- sorry -- is the suggestion to just note somewhere that the version of ROS 2 on the robot should match the version of ROS 2 on the device(s) that are communicating with it? If so, that's a reasonable suggestion. Do you think an "important" box on the "Releases" page would be a good place for that?

AndrewGerstenslager, 2024-03-18:
I'll look back at where I would think a good place will be. I'm busy today with a project but will be happy to offer more help tomorrow and later!